openwrt/luci #1029: Apply and save does not work

vic1126, 2017-02-21:
Hi
I add serial config using cbi,It displays correct, but when I input new content,  use save button, it does not work, any help? I am at LuCI 15.05-149-g0d8bbd2 Release.

local sys   = require "luci.sys"
local fs    = require "nixio.fs"
require("luci.config")

local m, s, o
m = Map("s2t", translate("Serial Server"), translate("Configure Serial Server"))
m:chain("luci")
s = m:section(TypedSection, "Serial_Server",translate("Serial Server"))
s.addremove = false
s.anonymous = true
s:tab("general",  translate("General Settings"))
o = s:taboption("general", Value, "comm_port", translate("Comm Port"))
function o.write(self, section, value)
   m.uci:set("s2t", "Serial_Server", "comm_port", "TEST")
end

return m

psyborg55, 2017-03-14:
i've noticed this also, tried once again and it applied the changes

bnelson61, 2017-04-03:
Did you try and add this to /etc/config/ucitrack?

jow-, 2018-07-03:
No response and no official LuCI code.

closed 2018-07-03 by jow-