openwrt/luci #1045: ubuntu build luCI fail

huanglshu, 2017-03-01:
When build uhttpd package, I find svn repository at URL 'svn://svn.open.wrt.org' can't use.so I git clone uhttpd src at URL 'git://git.openwrt.org/project/uhttpd.git',but I can find the Makefile? Where I can get uhttpd MakeFile?

hnyman, 2017-03-01:
You need to compile uhttpd using the cross-compiling toolchain from the main Openwrt build. 

git.openwrt.org is deprecated. If you want up-to-date Openwrt sources (including uhttpd package Makefile), clone from https://github.com/openwrt/openwrt.git

closed 2017-03-01 by hnyman

huanglshu, 2017-03-02:
I want to develop LuCI on MacOS, but I don't know how to set-up develop environment, Has any tutorial to learn？

hnyman, 2017-03-02:
wikis contain the advice that there is:
https://wiki.openwrt.org/about/toolchain
https://lede-project.org/docs/guide-developer/start

But you are largely on your own. You can ask question on the forums.