openwrt/luci #1052: Rounding issues with jsonc.stringify()

gbuchetet, 2017-03-04:
Hello,

Numbers are incorrectly dumped by the jsonc.stringify() function.

`require "luci.jsonc"

t = { a = 0.1 }
print(luci.jsonc.stringify(t))
print(t.a)`

The code above outputs:
`{"a":0.10000000000000001}
0.1`

I am expecting `{"a":0.1}` instead.

jow-, 2017-05-03:
Is this just a cosmetical issue or causing actual troubles for you? Technically speaking, `0.10000000000000001` is the correct representation of `0.1` in double precision. Any IEEE 754 compatible floating point implementation should treat `0.10000000000000001` and `0.1` exactly the same when using double precision values.

Example:

    root@jj:~# lua -lluci.jsonc -e 'print(0.1 == 0.10000000000000001)'
    true
    root@jj:~# lua -lluci.jsonc -e 'print(0.1 == 0.10000000000000002)'
    false

We could try to solve this particular problem by unconditionally truncating doubles to 14 digits, but then we loose precision in some cases.

jow-, 2020-02-23:
Timeout.

closed 2020-02-23 by jow-