openwrt/luci #1055: File upload API is not safe?

xm0625, 2017-03-05:
The Interface below
http://xxx.xxx.xxx.xxx/cgi-bin/luci/;stok=59d723686c9131e495c383d14ae072d8/admin/system/flashops
could be used to upload file to the RAM, although I don't have the auth(no cookie & wrong  stok).
I could easily use this to crash the devices with 64M RAM(703N, 720N). The CPU  Usage of luci process up to 90%+,and the free RAM  decreasing quickly cause the OOM of kernel.   
Check the rest ram and limit the post data size is not a good way.(concurrent requests)
Auth first?

cshoredaniel, 2017-03-07:
Maybe this would be better with a 'DummyFileUpload' which is like FileUpload except doesn't change any UCI config (similar to the RFC I closed for other widgets, for now due to availability, but is available in the PR history for those interested).

cshoredaniel, 2017-03-08:
I see the issue; it's because http/protocol.lua parses the HTTP body before authentication.  I'm not sure how much can actually be done about that; but at least if the 'memory fallback' were removed, the worst that could happen is /tmp could could get full (= 1/2 RAM).

cshoredaniel, 2017-03-08:
The issue is the that authtoken is a field called token that is returned as part of a multipart-mime body (by the client) so authentication can't happen until the part of the mime message with the token is parsed; for a file upload the problem is that the POST includes the uploaded file.

xm0625, 2017-03-08:
so, with this condition, if I want to auth before parse a multipart-mime body, use http-auth or cookie?(I want to write some restful apis  for luci,  neither http-auth nor cookie is friendly to Restful...)

xm0625, 2017-03-08:
what "auth" I mean is how to get some value to "auth" without  parse a multipart-mime body.

cshoredaniel, 2017-03-08:
CSRF complicates that issue; REST I think tends to not use multipart-mime but a single JSON body (although other formats are possible).  You might want to check out the JSON-RPC LuCI option - I don't know the details but it is designed for programmatic vs. user-based interaction.  I think the problem you're having at the moment is trying to use the user-facing interface to do REST, which tends not to work well (APIs tend not to be oriented to interactive human use and vice-versa); with AJAX the human interface is moved off the (in this case, CGI) server to the browser, so that human-facing and API usage look the same to the server.

cshoredaniel, 2017-03-08:
Also current branch of LuCI doesn't use stok in URI; that's an old release (maybe stable OpenWrt vs LEDE release which uses mostly trunk LuCI)?  In any event checkout out JSON-RPC LuCI package.

jow-, 2018-07-03:
MIME parsing, upload handling etc. have been heavily reworked in master so the original issues likely do not apply anymore.

closed 2018-07-03 by jow-