openwrt/luci #1056: LUCI slow down on LEDE 17.01 with 32 MB devices

Snotmann, 2017-03-05:
Hey guys,

i got a problem with luci as you could see here:
https://forum.lede-project.org/t/luci-slow-down-to-death-wr1043n-v1/1992/35

Maybe someone here got an idea how i could fix this ?!

Regards,
Snot

hnyman, 2017-03-05:
Not really. The same people here as there.
There is no real cure for a device having too little memory for the current kernel, base packages & LuCI.

Snotmann, 2017-03-05:
So tell my what i should do or how can i help that my issue get fixed ?
If you give me your address i could send you my router so you can have a look ?! ;)

glycoknob, 2017-05-08:
@Snotmann zram might help, but this opens another can of worms and it never worked reliable for me. What you can if you have a device with more than 4mb flash is lowering the squashfs block size i.e. to 64kb - this should speed up reading and you could try lowering `vm.min_free_kbytes = 1024` to something like 128 - but this might also cause problems - but this frees some memory - it was something around 168kb in cc 15.05  - you could also try reducing the squashfs fragment cache (see this patch: https://gitlab.bau-ha.us/weimarnetz/firmware/blob/master/patches/801-fragment-cache-size.patch) - this also should more memory to the page cache.

megapro17, 2017-05-10:
openwrt don't have that problem

megapro17, 2017-06-01:
Still no fix?

jow-, 2017-06-01:
There is nothing we can fix on the LuCI side of things. The code didn't really change that much since OpenWrt CC.

closed 2017-06-01 by jow-

megapro17, 2017-06-01:
okay, will make ticket on lede page

oferreiro, 2017-12-19:
I solved this issue disabling uhttp http_keepalive.
option http_keepalive '0'

fontanon, 2020-06-07:
> I solved similar issue after disable uhttp http_keepalive.
> option http_keepalive '0'

Thank you! Worked like a charm :)