openwrt/luci #1058: uhttpd ssl fails certificate check because weak methods available

olmari, 2017-03-08:
@jow- 

OpenWrt Designated Driver 50104 / LuCI Master (git-17.046.32957-dea880e)

Certificate with RSA key uhttpd (compiled with luci-ssl-openssl) offers insecure methods thus even having valid certificate it will fail on browsers that raises red flag with any weak component existing (like Chrome).

Main culprits are SSL/TLS compression	and RC4. Both should be disabled in order to not raise red flags (ssllabs analyze also confirms). So one should at least disable cipher suites TLS_RSA_WITH_RC4_128_MD5 (0x4)  and TLS_RSA_WITH_RC4_128_SHA (0x5), and not use ssl compression (enabling CRIME attack).

As a warning (but not a failure, yet at least) Perfect forward secrecy is also not supported, would be cool to have, but not red flag at this stage. As attachments screenshots from ssllabs test against openwrt uhttpd setup:
[image]
[image]
[image]

Also uhttpd doesn't seem to support certificates with EC-key? Tough that also isn't directly a show-stopper for now :)

hnyman, 2017-03-08:
Not related to Luci itself. The uhttpd init file (in Openwrt or LEDE main repo) creates the certificate using the included openssl library. Luci knows nothing about the certificate generation.

It is about the protocols/ciphers enabled in the Openssl library itself. There is no special version for uhttpd.

Do you get the same warnings for a certificate generated with LEDE's openssl library? In general it is a year ahead on development compared to stale Openwrt DD, but I haven't checked myself if the openssl cipher inclusion settings have been changed..

olmari, 2017-03-08:
I've created proper real certificate elsewhere and transferred into openwrt router afterwards (acme-tiny-dns to the win!), so no openwrt created certificates in this occasion, tough I don't see why would that really matter as it's http-server that agrees with key exchange and some other relating "transportation" stuff, not certificate itself.

That being said, I do have only little amount of knowledge on uhttpd itself, more apache and nginx where one can tell what cipher suites to support and what not to support in the http-service settings and they'll obey, no matter what ciphers and suites and other doohickeys openssl can support itself.

Extrapolating from what @hnyman said, there is no such method for uhttpd? (well, it _is_ 'micro' after all). And to continue that I would need to maybe choose what to support in openssl alltogether? And Now as I'm thinking of it, it is indeed not luci issue, but exactly a uhttpd issue really, so this is kind of in wrong place now I feel...

closed 2017-03-08 by olmari