openwrt/luci #1068: Drop invalid packets doesn't do anything

xrat, 2017-03-14:
This is my 1st bug report for LEDE. Please apologize my ineptness.

I am running LuCI lede-17.01 branch (git-17.051.53299-a100738) / LEDE Reboot 17.01.0 r3205-59508e3, specifically this [squashfs-factory.bin](
https://downloads.lede-project.org/releases/17.01.0/targets/ar71xx/generic/lede-17.01.0-r3205-59508e3-ar71xx-generic-tl-wr1043nd-v4-squashfs-factory.bin).

LuCI offers the option "Drop invalid packets" @ `/cgi-bin/luci/admin/network/firewall`. Starting off from default settings, the option is enabled, however, I cannot find a respective rule in `iptables -L`. When I uncheck the option in the web interface the line `option drop_invalid '0'` is added to `/etc/config/firewall`. When I check it the line is removed, but iptables rules don't change in either way.

Moreover, invalid packets are indeed not dropped according to my findings, documented in [NAT leakage on TL-WR1043ND v4 - LEDE Project Forum](https://forum.lede-project.org/t/nat-leakage-on-tl-wr1043nd-v4/1712 "Accessed 2017-03-14 12:30 +0100").

jow-, 2017-03-14:
Fixed in master and lede-17.01 branches. The NAT leakage needs a better fix as drop_invalid is no suitable solution since it causes valid IPv6 multicast traffic to get dropped.

closed 2017-03-14 by jow-

hnyman, 2017-03-14:
This might be due to changes in firewall defaults a while ago. The default used to be drop_invalid 1, but it was change to 0 by https://git.lede-project.org/?p=project/firewall3.git;a=commitdiff;h=b33f78371e7c7b6a131c2b6c01673cbd4b3c13d1;hp=6009904eb4dfe91e3d0b5cb96c4095232a303835

Might be that LuCI needs to be change to reflect the current defaults and to emit drop_invalid '1' into the config if needed. I haven't checked it myself, but in any case, @jow- should know best