openwrt/luci #1069: missing channel and bitrate info

psyborg55, 2017-03-14:
using ar9271 usb card in STA mode and having vif AP disabled will result in lack of channel and bitrate info that should display under Generic MAC80211 802.11bgn (radio2).

removing AP interface makes this informations shown

jow-, 2017-03-14:
Please provide the output of `iwinfo` and `ubus call network.wireless status`

psyborg55, 2017-03-15:
```
wlan0     ESSID: "Xiaomi_5_1"
          Access Point: 64:09:80:00:00:00
          Mode: Master  Channel: 48 (5.240 GHz)
          Tx-Power: 17 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: 14C3:7662 14C3:7662 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0

wlan0-1   ESSID: "Xiaomi_5_2"
          Access Point: 66:09:80:00:00:00
          Mode: Master  Channel: 48 (5.240 GHz)
          Tx-Power: 17 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: 14C3:7662 14C3:7662 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0

wlan0-2   ESSID: "Xiaomi_5_3"
          Access Point: 62:09:80:00:00:00
          Mode: Master  Channel: 48 (5.240 GHz)
          Tx-Power: 17 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: 14C3:7662 14C3:7662 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0

wlan0-3   ESSID: "Xiaomi_5_4"
          Access Point: 6E:09:80:00:00:00
          Mode: Master  Channel: 48 (5.240 GHz)
          Tx-Power: 17 dBm  Link Quality: 42/70
          Signal: -69 dBm  Noise: unknown
          Bit Rate: 300.0 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: 14C3:7662 14C3:7662 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0

wlan0-4   ESSID: "Xiaomi_5_5"
          Access Point: 6A:09:80:00:00:00
          Mode: Master  Channel: 48 (5.240 GHz)
          Tx-Power: 17 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: 14C3:7662 14C3:7662 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0

wlan1     ESSID: "Xiaomi_2_1"
          Access Point: 64:09:80:00:00:00
          Mode: Master  Channel: 13 (2.472 GHz)
          Tx-Power: 20 dBm  Link Quality: unknown/70
[... 262 lines cut ...]
			"txpower": 20,
			"country": "GB",
			"channel": "1"
		},
		"interfaces": [
			{
				"section": "@wifi-iface[10]",
				"ifname": "wlan2",
				"config": {
					"ssid": "link",
					"encryption": "psk2+ccmp",
					"key": "z471kXs9a5Cn2s3Vt8",
					"macaddr": "00:15:6D:6E:11:BF",
					"mode": "sta",
					"network": [
						"wwan"
					],
					"mode": "sta"
				}
			}
		]
	}
}

```

jow-, 2017-03-15:
Please also attach the JSON produced by http://192.168.1.1/cgi-bin/luci/admin/network/wireless_status/radio0.network1,radio0.network2,radio0.network3,radio0.network4,radio0.network5,radio1.network1,radio1.network2,radio1.network3,radio1.network4,radio1.network5,radio2.network1

psyborg55, 2017-03-16:
```

[{"ifname":"wlan0","disabled":false,"encryption":"WPA2 PSK (CCMP)","bssid":"64:09:80:00:00:00","mode":"Master","quality":0,"noise":0,"ssid":"Xiaomi_5_1",
"link":"\/cgi-bin\/luci\/admin\/network\/wireless\/radio0.network1","id":"radio0.network1","txpoweroff":0,
"device":{"device":"radio0","name":"Generic 802.11ac Wireless Controller (radio0)","up":true},"country":"GB","txpower":17,
"name":"Master \"Xiaomi_5_1\"","channel":56,"assoclist":[],"signal":0,"up":true,"frequency":"5.280"},
{"ifname":"wlan0-1","disabled":false,"encryption":"WPA2 PSK (CCMP)","bssid":"66:09:80:00:00:00","mode":"Master","quality":0,"noise":0,"ssid":"Xiaomi_5_2",
"link":"\/cgi-bin\/luci\/admin\/network\/wireless\/radio0.network2","id":"radio0.network2","txpoweroff":0,
"device":{"device":"radio0","name":"Generic 802.11ac Wireless Controller (radio0)","up":true},"country":"GB","txpower":17,
"name":"Master \"Xiaomi_5_2\"","channel":56,"assoclist":[],"signal":0,"up":true,"frequency":"5.280"},
{"ifname":"wlan0-2","disabled":false,"encryption":"WPA2 PSK (CCMP)","bssid":"62:09:80:00:00:00","mode":"Master","quality":0,"noise":0,"ssid":"Xiaomi_5_3",
"link":"\/cgi-bin\/luci\/admin\/network\/wireless\/radio0.network3","id":"radio0.network3","txpoweroff":0,
"device":{"device":"radio0","name":"Generic 802.11ac Wireless Controller (radio0)","up":true},"country":"GB","txpower":17,
"name":"Master \"Xiaomi_5_3\"","channel":56,"assoclist":[],"signal":0,"up":true,"frequency":"5.280"},
{"ifname":"wlan0-3","disabled":false,"encryption":"WPA2 PSK (CCMP)","bssid":"6E:09:80:00:00:00","mode":"Master","quality":58,"noise":0,"ssid":"Xiaomi_5_4",
"link":"\/cgi-bin\/luci\/admin\/network\/wireless\/radio0.network4","id":"radio0.network4","txpoweroff":0,
"device":{"device":"radio0","name":"Generic 802.11ac Wireless Controller (radio0)","up":true},"country":"GB","bitrate":243,"txpower":17,
"name":"Master \"Xiaomi_5_4\"","channel":56,"assoclist":{"2C:D0:5A:73:2D:C1":{"rx_short_gi":true,"rx_vht":false,"rx_mhz":40,"rx_40mhz":true,"tx_packets":161610,"tx_mhz":40,"rx_packets":141337,"rx_ht":true,"tx_mcs":14,"noise":0,"rx_mcs":15,"tx_rate":243000,"inactive":190,"tx_short_gi":false,"tx_40mhz":true,"tx_ht":true,"tx_vht":false,"rx_rate":300000,"signal":-69}},"signal":-69,"up":true,"frequency":"5.280"},
{"ifname":"wlan0-4","disabled":false,"encryption":"None","bssid":"6A:09:80:00:00:00","mode":"Master","quality":0,"noise":0,"ssid":"Xiaomi_5_5",
"link":"\/cgi-bin\/luci\/admin\/network\/wireless\/radio0.network5","id":"radio0.network5","txpoweroff":0,
"device":{"device":"radio0","name":"Generic 802.11ac Wireless Controller (radio0)","up":true},"country":"GB","txpower":17,
"name":"Master \"Xiaomi_5_5\"","channel":56,"assoclist":[],"signal":0,"up":true,"frequency":"5.280"},
{"ifname":"wlan1","disabled":false,"encryption":"WPA2 PSK (CCMP)","bssid":"64:09:80:00:00:00","mode":"Master","quality":0,"noise":0,"ssid":"Xiaomi_2_1",
"link":"\/cgi-bin\/luci\/admin\/network\/wireless\/radio1.network1","id":"radio1.network1","txpoweroff":0,
"device":{"device":"radio1","name":"Generic 802.11bgn Wireless Controller (radio1)","up":true},"country":"GB","txpower":20,
"name":"Master \"Xiaomi_2_1\"","channel":13,"assoclist":[],"signal":0,"up":true,"frequency":"2.472"},
{"ifname":"wlan1-1","disabled":false,"encryption":"WPA2 PSK (CCMP)","bssid":"64:09:80:00:00:00","mode":"Master","quality":0,"noise":0,"ssid":"Xiaomi_2_2",
"link":"\/cgi-bin\/luci\/admin\/network\/wireless\/radio1.network2","id":"radio1.network2","txpoweroff":0,
"device":{"device":"radio1","name":"Generic 802.11bgn Wireless Controller (radio1)","up":true},"country":"GB","txpower":20,
"name":"Master \"Xiaomi_2_2\"","channel":13,"assoclist":[],"signal":0,"up":true,"frequency":"2.472"},
{"ifname":"wlan1-2","disabled":false,"encryption":"WPA2 PSK (CCMP)","bssid":"64:09:80:00:00:00","mode":"Master","quality":0,"noise":0,"ssid":"Xiaomi_2_3",
"link":"\/cgi-bin\/luci\/admin\/network\/wireless\/radio1.network3","id":"radio1.network3","txpoweroff":0,
"device":{"device":"radio1","name":"Generic 802.11bgn Wireless Controller (radio1)","up":true},"country":"GB","txpower":20,
"name":"Master \"Xiaomi_2_3\"","channel":13,"assoclist":[],"signal":0,"up":true,"frequency":"2.472"},
{"ifname":"wlan1-3","disabled":false,"encryption":"WPA2 PSK (CCMP)","bssid":"64:09:80:00:00:00","mode":"Master","quality":0,"noise":0,"ssid":"Xiaomi_2_4",
"link":"\/cgi-bin\/luci\/admin\/network\/wireless\/radio1.network4","id":"radio1.network4","txpoweroff":0,
"device":{"device":"radio1","name":"Generic 802.11bgn Wireless Controller (radio1)","up":true},"country":"GB","txpower":20,
"name":"Master \"Xiaomi_2_4\"","channel":13,"assoclist":[],"signal":0,"up":true,"frequency":"2.472"},
{"ifname":"wlan1-4","disabled":false,"encryption":"None","bssid":"64:09:80:00:00:00","mode":"Master","quality":0,"noise":0,"ssid":"Xiaomi_2_5",
"link":"\/cgi-bin\/luci\/admin\/network\/wireless\/radio1.network5","id":"radio1.network5","txpoweroff":0,
"device":{"device":"radio1","name":"Generic 802.11bgn Wireless Controller (radio1)","up":true},"country":"GB","txpower":20,
"name":"Master \"Xiaomi_2_5\"","channel":13,"assoclist":[],"signal":0,"up":true,"frequency":"2.472"},
{"ifname":"wlan2","disabled":false,"encryption":"WPA2 PSK (CCMP)","bssid":"00:15:6D:6E:11:BF","mode":"Client","quality":70,"noise":0,"ssid":"link",
"link":"\/cgi-bin\/luci\/admin\/network\/wireless\/radio2.network1","id":"radio2.network1","txpoweroff":0,
"device":{"device":"radio2","name":"Generic 802.11bgn Wireless Controller (radio2)","up":true},"country":"GB","bitrate":54,"txpower":20,
"name":"Client \"link\"","channel":1,"assoclist":{"00:15:6D:6E:11:BF":{"noise":0,"rx_vht":false,"rx_mhz":20,"tx_ht":false,"tx_vht":false,"tx_rate":54000,"tx_packets":224276,"tx_mhz":20,"rx_packets":531432,"inactive":30,"rx_ht":false,"rx_rate":54000,"signal":-61}},"signal":-61,"up":true,"frequency":"2.412"}]

```

jow-, 2017-03-16:
Did you gather that data in the broken or the working state?

psyborg55, 2017-03-16:
in the broken state. STA enabled and connected, AP disabled.

psyborg55, 2017-03-28:
broken because of stupid mac80211 change introduced several years ago. fixed with this patch

```diff -pur a/net/mac80211/cfg.c b/net/mac80211/cfg.c
--- a/net/mac80211/cfg.c	2016-01-29 23:45:39.533154000 +0100
+++ b/net/mac80211/cfg.c	2016-01-30 01:03:16.367920000 +0100
@@ -540,7 +540,7 @@ static int ieee80211_set_monitor_channel
 			ret = ieee80211_vif_use_channel(sdata, chandef,
 					IEEE80211_CHANCTX_EXCLUSIVE);
 		}
-	} else if (local->open_count == local->monitors) {
+	} else {
 		local->_oper_chandef = *chandef;
 		ieee80211_hw_config(local, 0);
 	}
@@ -3231,9 +3231,7 @@ static int ieee80211_cfg_get_channel(str
 	if (chanctx_conf) {
 		*chandef = sdata->vif.bss_conf.chandef;
 		ret = 0;
-	} else if (local->open_count > 0 &&
-		   local->open_count == local->monitors &&
-		   sdata->vif.type == NL80211_IFTYPE_MONITOR) {
+	} else {
 		if (local->use_chanctx)
 			*chandef = local->monitor_chandef;
 		else
diff -pur a/net/mac80211/tx.c b/net/mac80211/tx.c
--- a/net/mac80211/tx.c	2016-01-15 13:00:10.000000000 +0100
+++ b/net/mac80211/tx.c	2016-01-30 01:05:48.032662000 +0100
@@ -798,9 +798,12 @@ ieee80211_tx_h_sequence(struct ieee80211
 	 * number, if we have no matching interface then we
 	 * neither assign one ourselves nor ask the driver to.
 	 */
-	if (unlikely(info->control.vif->type == NL80211_IFTYPE_MONITOR))
+	if (unlikely((info->flags & IEEE80211_TX_CTL_INJECTED) &&
+	   !(tx->sdata->u.mntr_flags & MONITOR_FLAG_COOK_FRAMES))) {
+		if (!ieee80211_has_morefrags(hdr->frame_control))
+			info->flags |= IEEE80211_TX_CTL_NO_ACK;
 		return TX_CONTINUE;
-
+	}
 	if (unlikely(ieee80211_is_ctl(hdr->frame_control)))
 		return TX_CONTINUE;
 
@@ -1661,7 +1664,9 @@ void ieee80211_xmit(struct ieee80211_sub
 		}
 	}
 
-	ieee80211_set_qos_hdr(sdata, skb);
+	if (likely(info->control.vif->type != NL80211_IFTYPE_MONITOR)) {
+		ieee80211_set_qos_hdr(sdata, skb);
+	}
 	ieee80211_tx(sdata, sta, skb, false);
 }
 
diff -pur a/net/wireless/chan.c b/net/wireless/chan.c
--- a/net/wireless/chan.c	2016-01-15 13:00:10.000000000 +0100
+++ b/net/wireless/chan.c	2016-01-30 01:06:03.448737000 +0100
@@ -857,8 +857,6 @@ int cfg80211_set_monitor_channel(struct 
 {
 	if (!rdev->ops->set_monitor_channel)
 		return -EOPNOTSUPP;
-	if (!cfg80211_has_monitors_only(rdev))
-		return -EBUSY;
 
 	return rdev_set_monitor_channel(rdev, chandef);
 }

jow-, 2020-02-23:
I assume this is fixed in master

closed 2020-02-23 by jow-