openwrt/luci #1070: luci-theme-material: there is no style for .cbi-input-invalid.

aa65535, 2017-03-15:
When the input data is invalid, it will be added class `.cbi-input-invalid`, but the theme did not give the corresponding style as a hint.

djhamzas, 2017-03-23:
sorry  i am a new in luci I want to start developing LUCI ,  How can i Get started , is there an editor for that 
what tools

djhamzas, 2017-03-23:
Any Help

hnyman, 2017-03-23:
> sorry i am a new in luci I want to start developing LUCI , How can i Get started , is there an editor for that
what tools

Luci source code is just text files. You can edit those with any text editor (gedit, nano, emacs, whatever...)
And there is no visual integrated development environment or anything like that. So the learning curve is pretty high.

(and please, ask this kind of questions in the Openwrt / LEDE forums, as your questin has nothing to do with this isue that you hijacked)

commit 9b20f7ac (2018-11-05): luci-theme-material: add invalid input class

Fixes #1070.
closed 2018-11-05 by jow-