openwrt/luci #1073: LuCI Admin: Network DHCP-and-DNS unconditionally looks for dnsmasq

EricLuehrsen, 2017-03-18:
The admin page under menu "Network->DHCP and DNS" appears to look unconditionally for dnsmasq. This is noticed with dnsmasq not installed. odhcpd is standalone with or out alternate DNS server in use (Unbound, Bind). Everything appears to work out anyway. There may be a latent bug worth looking at.

`Sat Mar 18 13:36:06 2017 daemon.err uhttpd[1671]: /usr/bin/ldd: cannot load /usr/sbin/dnsmasq: No such file or directory`

cc: @hnyman, @cshoredaniel

cshoredaniel, 2017-03-18:
@EricLuehrsen That's the test for if the DNSSEC options should be enabled - if that applies to unbound and/or bind as well then the check needs to be modified (that is if they support those UCI options).  Suppressing the error message is really all that's needed since the test for linking to libnettle(?) fails if the binary dnsmasq isn't even present, which means the options aren't shown).

EricLuehrsen, 2017-03-19:
Both applications are compiled with DNSSEC and no option otherwise. BIND doesn't have UCI. Unbound UCI is under **services**.  DNS is a service which is not necessary to connecting the local network. DHCP or such relay is necessary. The DHCP option for DNS can direct to the ISP (long ago this was the only way). I get that dnsmasq covers the 80% use case of names n' numbers all in one, but LuCI layout is a bit odd from from a function-block design perspective. Anyway, I am getting off track. (1) if it does no harm, then I guess to don't really care. (2) as pattern practice, should we test if something is present before making that something an argument to an external call?

cshoredaniel, 2017-03-19:
@EricLuehrsen 2) would probably be preferrable; I just hadn't really thought about dnsmasq not being present in a case where you'd use the DNS app.

EricLuehrsen, 2017-03-19:
It seems its linked to anywhere you would manage DHCP (odhcpd only) such as fixed host addresses.

cshoredaniel, 2017-03-19:
@EricLuehrsen probably the best thing to do is modify the link check function so only one update needs to be made

jow-, 2020-02-23:
Master uses feature flags to enable/disable options and the entire view is removed if no `/etc/config/dhcp` exists.

closed 2020-02-23 by jow-