openwrt/luci #1075: Allow network-interface type PPPoE to set custom mac address

Strykar, 2017-03-19:
Currently there is no option to set a custom MAC address if the ethernet interface is configured for PPPoE.
Many ISPs will block the connection unless it comes from a registered MAC.

Setting `option macaddr` to the appropriate ethx interface in `/etc/config/network` does not appear to fit it.
`ifconfig ethx` still shows the original MAC address.

Tested on LEDE Reboot (SNAPSHOT, r3582-8873474)

qiuzi, 2017-04-07:
config device 'wan_dev'
option macaddr

hnyman, 2017-04-07:
LuCI is just the GUI where we can set network options in /etc/config/network, but LuCI has no role in the actual Linux network config handling. It is mainly the "netifd" daemon in the main sources that takes care of the interface config based on /etc/config/network.

But the underlying core issue (and the proposed config fix suggested by @qiuzi ) sounds related to issue #1100 from @qiuzi . Like I commented there, the wan_dev "device" section itself is due to a commit in Feb 2016 (still in Openwrt times): https://github.com/lede-project/source/commit/f70c8ba

@nbd168 @jow- 
There seems to be confusion about the correct options to be used for MAC override in /etc/config/network etc. for pppoe. Commit https://github.com/lede-project/source/commit/f70c8ba from nbd introduced the device section usage for special MAC cases, and I wonder if that is now affecting users on certain target platforms negatively. Issue #1100 from @qiuzi is rather similar. 

I am not sure if "device" section config is supported in LuCI at all, but I think that it is a side issue.

The underlying core issue is the correct option for netifd etc.  Is the "interface" section's macaddr option used, or does the possible "device" section macaddr override that?  This whole issue should maybe be in the LEDE main bug tracker instead of LuCI. (except for that part of LuCI support for "device" sections)

barisgulten, 2017-10-15:
Hi, what about destination mac address in pppoed packets. 
My operators router directly searching specific mac address on pppoed packets. 
Can I change this area ?
I am using Lede LuCI by the way. I managed to change operator provided routers mac address to my Lede router. 
According to wireshark logs I need to change destination mac address in order to pass operator, I believe.

maxxzaxx, 2018-09-09:
Ithrough SSH I edited /etc/config/network
And did the following

config interface 'wan'
        option ifname 'eth0'
        option _orig_ifname 'eth0'
        option _orig_bridge 'false'
        option proto 'pppoe'
        option password 'xxxxxxxx'
        option metric '1'
        option macaddr 'xx:xx:xx:xx:xx:xx'
        option username 'xxxxxx'

Am a novice but had luck this time.
Cheers.

feckert, 2019-08-11:
This is solved

closed 2019-08-11 by feckert