openwrt/luci #1086: VLAN config not written correctly

dgilmour77, 2017-03-27:
Summary: LuCI modifies the VLAN config when saving even if user makes no changes, likely resulting in bricked router.

Reported on LEDE 17.01.0

Starting with a firstboot config for the Linksys EA8500, which predefines two VLANs, go to LuCI page for switch configuration.   The VLANs correctly show tagged connections to the CPU.

Without changing anything, click "save and apply".  The network configuration is updated, but the CPU connections are omitted in VLANs 1 and 2.  The following show the original, correct config and the modified, broken config.  As you can see, the "0t" is deleted from the option ports line.  Unless the user has a backdoor wireless interface enabled for dropbear, this bricks the router and requires opening up again, and running firstboot from the serial console to restore operation of any of the Ethernet ports.

ORIGINAL config/network
...
config switch_vlan
  option device 'switch0'
  option vlan '1'
  option ports '1 2 3 4 0t'

config switch_vlan
  option device 'switch0'
  option vlan '2'
  option ports '5 0t
...

AFTER SAVED BY LUCI WITH NO USER CHANGES:
...
config switch vlan
  option device 'switch0'
  option vlan '1'
  option ports '1 2 3 4'                   <=NOTE '0t" is dropped from this config line!!
  option vid '1'

config switch vlan
  option device 'switch0'
  option vlan '2'
  option ports '5'
  option vid '2'
...

jow-, 2017-05-03:
Please provide the contents of the `/etc/board.json` file on your device.

jow-, 2017-05-03:
Fixed with https://git.lede-project.org/3c3f7cf and https://git.lede-project.org/bc0de27

closed 2017-05-03 by jow-