openwrt/luci #1087: probably latest changes luci-app-adblock - some erros to luci working

build000, 2017-03-28:
```
Błąd przetwarzania XML: nieprawidłowo sformowany Obszar: https://192.168.4.1/cgi-bin/luci/admin/services Numer wiersza: 122, kolumna 175:
<input name="runtime" id="runtime" type="text" class="cbi-input-text" style="border: none; box-shadow: none; background-color: #ffffff; color: #0069d6;" value="n/a" disabled />
```

hnyman, 2017-03-28:
Sorry, Polish is not my strong suit :-(

But how does the error manifest itself? just the error message about somthing invalid? 
and exactly on which page?

cc to author: @dibdot

dibdot, 2017-03-28:
OK, that's easy - I'll send a fix later today.

dibdot, 2017-03-28:
#1088

build000, 2017-03-28:
Confirm: this commit https://github.com/openwrt/luci/commit/3dc08a354b4f5212fdaf54764729a025a14167cf fixing problem .
luci-app-adblock..."new" working...hehe.

closed 2017-03-29 by hnyman