openwrt/luci #1100: Override MAC address Incorrect

qiuzi, 2017-04-07:
Wrongly changed to get "config interface 'wan'"
Option macaddr

The correct location should be in the "config device 'wan_dev'"
Option macaddr

hnyman, 2017-04-07:
What LuCI page/app you are talking about?

qiuzi, 2017-04-07:
@hnyman /cgi-bin/luci/admin/network/network

qiuzi, 2017-04-07:
[image]

hnyman, 2017-04-07:
What is your use case with more detail? 

In the normal situation where WAN has DHCP protocol, that option is not even visible. If WAN uses static address, this option is visble. But there is no wan_dev section in the network config if wan is related to the normal fixed wan port.

qiuzi, 2017-04-07:
@hnyman But the most openwrt source code ’/etc/config/network‘ file does have "config device 'wan_dev'"
[image]

hnyman, 2017-04-07:
> But the most openwrt source code ’/etc/config/network‘ file does have "config device 'wan_dev'"

No, it doesn't, at least usually. The typical /etc/config/network contains no "device" sections.
Below is my /etc/config/network as an example.

Your config is likely related to specific hardware and/or protocols, when e.g. the same physical interface is used for lan or wan, or something like that, so that the "device" section is needed. So please explain more closely your device, target platform etc., as it sounds like your problem is related to a niche case.

But in any case, the config item you see in LuCI is for overriding the macaddr of an interface, not the of an device. 

The option to handle device options should be separate, but I must admit that I am not sure if those "device" sections are even visible in LuCI. 

```
root@LEDE:~# cat /etc/config/network

config interface 'loopback'
        option ifname 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd1b:7654:3210::/48'

config interface 'lan'
        option type 'bridge'
        option ifname 'eth0.1'
        option proto 'static'
        option ipaddr '192.168.1.1'
        option netmask '255.255.255.0'
        option ip6assign '60'

config interface 'wan'
        option ifname 'eth1.2'
        option proto 'dhcp'

config interface 'wan6'
        option ifname 'eth1.2'
        option proto 'dhcpv6'

config switch
        option name 'switch0'
        option reset '1'
        option enable_vlan '1'

config switch_vlan
        option device 'switch0'
        option vlan '1'
        option ports '0 1 2 3 5t'
        option vid '1'

config switch_vlan
        option device 'switch0'
        option vlan '2'
        option ports '4 6t'
        option vid '2'

```

qiuzi, 2017-04-07:
@hnyman  Lenovo Y1 
'/bin/config_generate' This script defines “set network. $ 1_dev = 'device'”

qiuzi, 2017-04-07:
Thus affecting the entire mac address legitimate distribution

hnyman, 2017-04-07:
Hmm. Y1 is apparently a ramips device.
and there seems to be logic for initial settings of the MACs for lan and wan:
https://github.com/lede-project/source/blob/master/target/linux/ramips/base-files/etc/board.d/02_network#L318

But the wan_dev section itself is due to a commit in Feb 2016 (still in Openwrt times):
https://github.com/lede-project/source/commit/f70c8ba1980585e3f80499bba5a043e0b4940e4b

Like I said, I am not sure if there is support for configuring those "device" sections in LuCI. The option you are using now, is for interface section config. It is possible that you need to edit it by hand in the uci config file.

Is it not working for you that you change just the MAC of the wan interface?

qiuzi, 2017-04-07:
Yes! Existing luci source code on the current lede system can not modify the MAC
LEDE has not processed the "macaddr" parameter in the "interface" form

svoboda-jan, 2018-02-10:
Hello, I came across the same issue with fresh LEDE `17.01.4`, I had to set the `macaddr` manually:

Relevant config:
```
config interface 'lan'
	option type 'bridge'
	option ifname 'eth0.1'
	option proto 'static'
	option netmask '255.255.255.0'
	option ip6assign '60'
	option ipaddr '192.168.1.1'

config device 'lan_dev'
	option name 'eth0.1'
	option macaddr 'AA:AA:AA:AA:AA:01'

config interface 'wan'
	option ifname 'eth0.2'
	option proto 'dhcp'
	option peerdns '0'
	option dns '208.67.222.222 208.67.220.220'
	# set by LuCI, does not take effect:
	option macaddr 'CC:CC:CC:CC:CC:CC'

config device 'wan_dev'
	option name 'eth0.2'
	#option macaddr 'AA:AA:AA:AA:AA:02'
	# manually set by editing /etc/config/network, this works:
	option macaddr 'CC:CC:CC:CC:CC:CC'

config switch
	option name 'switch0'
	option reset '1'
	option enable_vlan '1'

config switch_vlan
	option device 'switch0'
	option vlan '1'
	option ports '0 1 2 3 6t'

config switch_vlan
	option device 'switch0'
	option vlan '2'
	option ports '4 6t'
```

ptipti23, 2019-02-16:
Override MAC still not working...

svoboda-jan, 2019-02-16:
As per my testing, this appears to be an issue only on certain devices with `*_dev` sections.

A workaround is to add `option macaddr 'xx...'` to the `config device 'something_dev'` section.

ptipti23, 2019-02-16:
> As per my testing, this appears to be an issue only on certain devices with `*_dev` sections.
> 
> A workaround is to add `option macaddr 'xx...'` to the `config device 'something_dev'` section.

Do you suggest simply deleting this section from the settings file? Is it really so hard to make it work?

svoboda-jan, 2019-02-16:
No, my suggestion is adding `option macaddr 'xx...'` to the section which ends which `_dev`.

ptipti23, 2019-02-16:
I have been doing this for a year now. How to stop doing this?

svoboda-jan, 2019-02-16:
Sorry, can't assist with that, was just posting data and a workaround that might be helpful for developers/users.

sophana, 2019-08-22:
I think this issue should be solved. It took me some time to figure out the cause of the problem.
Luci should check that shen setting mac address for interface XXX, then if XXX_dev exists, it should set the mac address on XXX_dev

MingdaHIT, 2019-10-20:
I got the sampe problem, and fix it by add option macaddr 'xx:xx:xx:xx:xx" in the config device 'wan_dev'. Like @svoboda-jan comment on 10 Feb 2018.

wonson, 2020-02-25:
@sophana confirmed device macaddr problem still exist in 2020. 
As @MingdaHIT , I use @svoboda-jan 's workaround to solve this.

I am using "LuCI openwrt-19.07 branch (git-20.029.45734-adbbd5c) / OpenWrt 19.07.1 r10911-c155900f66 " bundled in openWRT. 
Is it new enough to include your fix?

or this issue should be reopened?

wonson, 2020-02-25:
@sophana confirmed device macaddr problem still exists in 2020.
As @MingdaHIT , I use @svoboda-jan 's workaround to solve this.

I am using "LuCI openwrt-19.07 branch (git-20.029.45734-adbbd5c) / OpenWrt 19.07.1 r10911-c155900f66 " bundled in openWRT.

ptipti23, 2020-02-25:
There are new problems with new versions of OpenWRT, such as the strict requirement for the device’s memory, both the opervative and the physical. In general, to put it mildly, SoC in dir-300NRUB7 is no longer supported. I tried to collect new firmware, everything is useless. The only and latest firmware that works flawlessly is openwrt-18.06 branch (git-19.353.67991-868abc1) / OpenWrt 18.06.2 r7676-cddd7b4c77. I also collected it. If necessary, I can drop the download link. I’m already thinking about buying a new router

wonson, 2020-02-27:
http://downloads.openwrt.org/releases/18.06.2/targets/
@ptipti23 this version works?

DaLassOnline, 2020-07-30:
Having the same problem on my Engenius ENS202EXT. the option to change the MAC on the interface was working perfect on 18.06.4, but since I upgraded to 19.07.3, it doesn't work anymore. 

The interface I am trying to change is the trm_wwan interface created by Travelmate.

ghid1, 2020-09-19:
Using OpenWrt 19.07.4. Override MAC address doesn't work for WAN.

Nuck-TH, 2020-09-30:
Yeah, not working either on 19.07.4 on WG3526.
Override is set on interface instead of device, which doesn't work.
Maybe there should be option that if checked will set mac override on device(and it would be shown on all corresponding interfaces), and when unchecked - set per interface like now.

Str1ker17, 2020-11-05:
Override MAC address doesn't work for WAN on TP-Link Archer C20 v1 too.

Software used:
Powered by LuCI openwrt-19.07 branch (git-20.247.75781-0d0ab01) / OpenWrt 19.07.4 r11208-ce6496d796

fumantsu, 2021-01-31:
Override also doesn't work for OpenWrt SNAPSHOT r15642-eb11cd9ea3 / LuCI Master git-21.029.67968-af1f961
There is a valid case on that. Plenty of ISPs are locking the MAC on their modems/systems. If there is any change of router you need either to do an override or call the support to ask for change the MAC they have assign. I guessing that all we agree that the first option is easy and faster.

Str1ker17, 2021-02-04:
@fumantsu I'm not sure I understand what do you want to say. Do you want to say that overriding MAC on router through LuCI interface is useless?

fumantsu, 2021-02-04:
As the others said it is not working at all and needs to edit the files and
in comparison of other opinions which I read it is very useful, sometimes
necessary.

rsxxxx, 2021-02-04:
Somebody else linked me to this thread because I'm having the same problem with my NanoPi R4S using quintus-lab's build. LAN can be easily changed but WAN/WAN6 are unable to.

wonson, 2021-03-16:
Can we assume that if "wan_dev" exists, it has to edit the "wan_dev" properties?

jow-, 2021-03-18:
Fixed via faad7464a8ed88a251c621dc92e8cbfb146bcd7f - device specific properties are now transparently written into the proper corresponding `config device` section.

closed 2021-03-18 by jow-