openwrt/luci #1101: The wireless client signal strength is not displayed

Microdrive, 2017-04-07:
I think I found an error in the file luci-mod-admin-full/luasrc/view/admin_status/index.htm

 for( var i = 0; i < assoclist.length; i++ )
                                {
                                        var tr = ac.rows[0].parentNode.insertRow(-1);
                                                tr.className = 'cbi-section-table-row cbi-rowstyle-' + (1 + (i % 2));

                                        var icon;
                                        var q = (-1 * (assoclist[i].noise - assoclist[i].signal)) / 5;
                                        if (q < 1)
                                                icon = "<%=resource%>/icons/signal-0.png";
                                        else if (q < 2)
                                                icon = "<%=resource%>/icons/signal-0-25.png";
                                        else if (q < 3)
                                                icon = "<%=resource%>/icons/signal-25-50.png";
                                        else if (q < 4)
                                                icon = "<%=resource%>/icons/signal-50-75.png";
                                        else
                                                icon = "<%=resource%>/icons/signal-75-100.png";

The value of the variable assoclist [i] .signal is always less than 0, the value of the variable assoclist [i] .noise can take 0, then the variable q will always be less than 0 !!!
In this case, the variable "icon" is always equal to "/luci-static/resources/icons/signal-0.png"

[image]

bobmseagithub, 2017-07-16:
Confirmed on LEDE Reboot 17.01.2 r3435-65eec8bd5f, and LEDE Reboot 17.01.2 r3435-65eec8bd5f.

The existing code 
  var q = (-1 * (assoclist[i].noise - assoclist[i].signal)) / 5;
depends on the subtraction of two negatives being a positive.
However, sometimes noise comes into this code as 0.  This causes the subtraction result to be negative value, which selects the wrong icon.

Changing the formula from a multiply by -1, to a Math.abs() ensures that the result is always a positive value:
  var q = Math.abs((assoclist[i].noise - assoclist[i].signal)) / 5;

Why the noise is sometimes 0, which seems unlikely, is outside of the scope of this code.

feckert, 2019-09-13:
Works in master for me

closed 2019-09-13 by feckert