openwrt/luci #1105: luci-app-statistics -> error on installation

camelcamro, 2017-04-10:
#opkg update
opkg install luci-app-statistics –force-reinstall

##2. Install desired plugins (60 KB)
#opkg list | grep collectd-mod –force-reinstall

in logread - i see:

Sun Apr  9 12:43:34 2017 daemon.err uhttpd[2773]: /usr/lib/lua/luci/statistics/datatree.lua:16: attempt to index upvalue 'sections' (a nil value)
Sun Apr  9 12:43:34 2017 daemon.err uhttpd[2773]: stack traceback:
Sun Apr  9 12:43:34 2017 daemon.err uhttpd[2773]:       /usr/lib/lua/luci/statistics/datatree.lua:16: in function '__init__'
Sun Apr  9 12:43:34 2017 daemon.err uhttpd[2773]:       /usr/lib/lua/luci/util.lua:64: in function 'Instance'
Sun Apr  9 12:43:34 2017 daemon.err uhttpd[2773]:       .../luci/controller/luci_statistics/luci_statistics.lua:107: in function 'v'
Sun Apr  9 12:43:34 2017 daemon.err uhttpd[2773]:       /usr/lib/lua/luci/dispatcher.lua:563: in function 'createtree'
Sun Apr  9 12:43:34 2017 daemon.err uhttpd[2773]:       /usr/lib/lua/luci/dispatcher.lua:220: in function 'dispatch'
Sun Apr  9 12:43:34 2017 daemon.err uhttpd[2773]:       /usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>
Sun Apr  9 12:43:34 2017 daemon.err uhttpd[2773]: /usr/lib/lua/luci/statistics/datatree.lua:16: attempt to index upvalue 'sections' (a nil value)
Sun Apr  9 12:43:34 2017 daemon.err uhttpd[2773]: stack traceback:
Sun Apr  9 12:43:34 2017 daemon.err uhttpd[2773]:       /usr/lib/lua/luci/statistics/datatree.lua:16: in function '__init__'
Sun Apr  9 12:43:34 2017 daemon.err uhttpd[2773]:       /usr/lib/lua/luci/util.lua:64: in function 'Instance'
Sun Apr  9 12:43:34 2017 daemon.err uhttpd[2773]:       .../luci/controller/luci_statistics/luci_statistics.lua:107: in function 'v'
Sun Apr  9 12:43:34 2017 daemon.err uhttpd[2773]:       /usr/lib/lua/luci/dispatcher.lua:563: in function 'createtree'
Sun Apr  9 12:43:34 2017 daemon.err uhttpd[2773]:       /usr/lib/lua/luci/dispatcher.lua:220: in function 'dispatch'
Sun Apr  9 12:43:34 2017 daemon.err uhttpd[2773]:       /usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>
Sun Apr  9 12:43:34 2017 daemon.err uhttpd[2773]: /usr/lib/lua/luci/statistics/datatree.lua:16: attempt to index upvalue 'sections' (a nil value)
Sun Apr  9 12:43:34 2017 daemon.err uhttpd[2773]: stack traceback:
Sun Apr  9 12:43:34 2017 daemon.err uhttpd[2773]:       /usr/lib/lua/luci/statistics/datatree.lua:16: in function '__init__'
Sun Apr  9 12:43:34 2017 daemon.err uhttpd[2773]:       /usr/lib/lua/luci/util.lua:64: in function 'Instance'
Sun Apr  9 12:43:34 2017 daemon.err uhttpd[2773]:       .../luci/controller/luci_statistics/luci_statistics.lua:107: in function 'v'
Sun Apr  9 12:43:34 2017 daemon.err uhttpd[2773]:       /usr/lib/lua/luci/dispatcher.lua:563: in function 'createtree'
Sun Apr  9 12:43:34 2017 daemon.err uhttpd[2773]:       /usr/lib/lua/luci/dispatcher.lua:220: in function 'dispatch'
Sun Apr  9 12:43:34 2017 daemon.err uhttpd[2773]:       /usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>

hnyman, 2017-04-10:
Most likely you have a faulty or missing /etc/config/luci_statistics. When you reinstall a package, opkg does not overwrite existing config files, but places the new default config as <filename>-opkg. So a possible problematic old config file is still in use.

You need to replace it with the new copy that opkg placed there:
```
cd /etc/config
cp luci_statistics-opkg luci_statistics
```

-----

while looking into this, I noticed an actual bug related to opkg downloads & automatics hashes in package names. I fail to reinstall luci-app-statistics to my own 17.01 system, as there has originally been a self-built package that I have now removed. But my buildhost has newer git than LEDE buildbot, so the package name of my own build has slightly longer default git hash and that does not match and opkg tries to download the "wrong" luci-app-statistics when installing.

```
root@OpenWrt2:/etc/config# opkg list luci-app-st*
luci-app-statistics - git-17.073.42825-b47a21f-1 - LuCI Statistics Application
luci-app-statistics - git-17.073.42825-b47a21fbb-1

root@OpenWrt2:/etc/config# opkg install luci-app-statistics
Installing luci-app-statistics (git-17.073.42825-b47a21fbb-1) to root...
Collected errors:
 * opkg_download_pkg: Package luci-app-statistics is not available from any configured src.
 * opkg_install_pkg: Failed to download luci-app-statistics. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package luci-app-statistics.
```

hnyman, 2017-04-10:
@jow- 
This might interest you.
I am still baffled about the underlying reason.

Like shown above, I am unable to re-install luci-app-statistics to **one** 17.01 router. I can opkg install it to another 17.01 router. 

The reason seems to be varying length of git commit hash in package name. In my self-built ar71xx package it is b47a21fbb, while buildbot has it b47a21f. As my build has have had the package preinstalled, the wrong name is still found in /usr/lib/opkg/status, opkg seems to prefer that, and the reinstall fails like above.
```
Package: luci-app-statistics
Version: git-17.073.42825-b47a21fbb-1
Depends: libc, collectd, rrdtool1, collectd-mod-rrdtool, collectd-mod-iwinfo, collectd-mod-cpu, collectd-mod-memory, collectd
-mod-interface, collectd-mod-load, collectd-mod-network
Status: install user not-installed
Architecture: all
Conffiles:
 /etc/config/luci_statistics e7eb298f447a39f8020657f2638be5ffbc60ba8754db85716d63e738fd61d3b1
Installed-Time: 1491763622
```
I first thought that this might be related to me having a much newer git version than the buildbot, but based on further investigation it is not that simple...

The really baffling part here is that for my ipq806x build on the same buildhost the git hash gets again used as 7 chars (like in buildbot) when git is used from console.  Both repos should have the same branches etc.:

```
perus@ub1610:/Openwrt/lede1701/feeds/luci$ cd /Openwrt/r7800-lede1701/feeds/luci

perus@ub1610:/Openwrt/r7800-lede1701/feeds/luci$ git log --oneline | head -n 2
b47a21f luci-app-firewall: adjust drop_invalid default (#1068)
a5191ef i18n: sync translations

perus@ub1610:/Openwrt/r7800-lede1701/feeds/luci$ git branch -a
* lede-17.01
  remotes/origin/lede-17.01

perus@ub1610:/Openwrt/r7800-lede1701/feeds/luci$ cd /Openwrt/lede1701/feeds/luci

perus@ub1610:/Openwrt/lede1701/feeds/luci$ git log --oneline | head -n 2
b47a21fbb luci-app-firewall: adjust drop_invalid default (#1068)
a5191efc7 i18n: sync translations

perus@ub1610:/Openwrt/lede1701/feeds/luci$ git branch -a
* lede-17.01
  remotes/origin/lede-17.01
```

So the interesting part here is that using the automatic git hash length in package name may vary even on the same buildhost and may lead into strange problems with opkg.

I am wondering 
* if we should change the hash logic to use a defined length hash (instead of relying on git automatics). something like   `git log -1 --format="%ct %h" --abbrev=10`
* if opkg should be changed to believe the downloaded package info file about the exact revision to be downloaded.

hnyman, 2017-04-10:
@jow- 
Additional interesting part is that the local version info from /usr/lib/opkg/status seems to override even an explicit filename to be downloaded. Note how at the installation stage the longer 9-char hash pops up:

```
root@OpenWrt2:/usr/lib/opkg# opkg install http://downloads.lede-project.org/releases/17.01-SNAPSHOT/packages/mips_24kc/luci/luci-app-statistics_git-17.073.42825-b47a21f-1_all.ipk
Downloading http://downloads.lede-project.org/releases/17.01-SNAPSHOT/packages/mips_24kc/luci/luci-app-statistics_git-17.073.42825-b47a21f-1_all.ipk
Installing luci-app-statistics (git-17.073.42825-b47a21fbb-1) to root...
Collected errors:
 * opkg_download_pkg: Package luci-app-statistics is not available from any configured src.
 * opkg_install_pkg: Failed to download luci-app-statistics. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package luci-app-statistics.

root@OpenWrt2:/usr/lib/opkg# opkg -v
opkg version 1d0263bb40e3c099501fc1f2431907636230e7f2 (2017-03-23)

(17.01-SNAPSHOT, r3305-c6e79980b8)
```

hnyman, 2017-05-10:
This missing/faulty luci_statistics file causing lockout from LuCI shold now be fixed with https://github.com/openwrt/luci/commit/d04f667383ec2894a5ecefde7ad9284d07563a67

closed 2017-05-10 by hnyman