openwrt/luci #1116: luci-admin-network: error executing page load

EricLuehrsen, 2017-04-17:
Well this is new. Haven't seen LuCI fail in a long time.
```
/usr/lib/lua/luci/dispatcher.lua:460: Failed to execute arcombine dispatcher target for entry '/admin/network/wireless/radio1.network1'.
The called action terminated with an exception:
/usr/lib/lua/luci/sys.lua:237: bad argument #1 to 'getnameinfo' (string expected, got nil)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:460: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>
```

EricLuehrsen, 2017-06-02:
I cannot figure out why this occurs. All getnameinfo() use nil as an argument. From 237.
```
function net.mac_hints(callback)
	if callback then
		_nethints(1, function(mac, v4, v6, name)
			name = name or nixio.getnameinfo(v4 or v6, nil, 100) or v4
			if name and name ~= mac then
				callback(mac, name or nixio.getnameinfo(v4 or v6, nil, 100) or v4)
			end
		end)
	else
```

EricLuehrsen, 2017-06-02:
@jow- 
I cannot figure out why this occurs. All getnameinfo() use nil as an argument. From 237.
```
function net.mac_hints(callback)
	if callback then
		_nethints(1, function(mac, v4, v6, name)
			name = name or nixio.getnameinfo(v4 or v6, nil, 100) or v4
			if name and name ~= mac then
				callback(mac, name or nixio.getnameinfo(v4 or v6, nil, 100) or v4)
			end
		end)
	else
```

jow-, 2017-06-02:
Seems like both "v4" and "v6" nil. We should probaby fix `mac_hints()` to only produce records which contain at least one IP.

jow-, 2019-09-23:
The `mac_hints()` code is not used by the interface config page anymore in master.

closed 2019-09-23 by jow-