openwrt/luci #1117: Dynamic DNS on the status overview page shows "Config error" under Network.

jebrossard, 2017-04-21:
LEDE Reboot 17.01.1 r3316-7eb58cf109 / LuCI lede-17.01 branch (git-17.100.70571-29fabe2)

After ADDING a ddns service in LuCI the ddns updating works just fine, but shows "Config error" under Network column on status overview.  This is because these options are missing from the newly added service's config:

        option interface
        option ip_source
        option ip_network

Once those options are manually added to the new ddns service's configuration, the "config error" goes away.  Please see the attached document.

[config ddns error.docx](https://github.com/openwrt/luci/files/947943/config.ddns.error.docx)

Tony-il-Capo, 2017-04-26:
Same problem, but seems it's already fixed. Good job!

closed 2017-05-03 by jow-

jebrossard, 2017-06-11:
This issue still exists with LEDE 17.01.2