openwrt/luci #1127: luc-app-adblock (adblock-2.6.1) - xml parsing error (theme openwrt)

build000, 2017-04-26:
system/luci version: LEDE Reboot SNAPSHOT r4042-e5bbead / LuCI Master (git-17.116.42736-415b99d)
theme: openwrt (only to this theme - bootstrap and material working fain)
page: https://"address"/cgi-bin/luci/admin/services
error:
```
XML Parsing Error: not well-formed Location: https://"address"/cgi-bin/luci/admin/services Line Number 1033, Column 84:
				Enable memory intense overall sort / duplicate removal on low memory devices (< 64 MB RAM)
---------------------------------------------------------------------------------------------------------------^
```

hnyman, 2017-04-26:
@dibdot 

Openwrt theme has the flags for strict XML/HTML formatting. I guess that the "<" in unquoted format is against that.

build000, 2017-04-27:
@dibdot 
confirm:
this PR https://github.com/openwrt/luci/pull/1128 fixing problem.

PR #1128 (2017-04-27): luci-app-adblock: fix markup
this fixes #1127
closed 2017-04-27 by hnyman