openwrt/luci #1129: Firewall-General: Use more clear generic Router descriptions instead of iptables descriptions

LipkeGu, 2017-04-27:
When creating Zones, we have the ability to create Zones and specify MSS-Fix and Masquerade, we should use more clear generic Router descriptions instead of iptables descriptions. Something like:

Masquerade -> NAT (Tooltip: "Enable Network Address Translation" for this Zone")
MSS-Fix -> MTU Discovery ("Adjust the Maximum Sequence Size for this Zone")

Newcomers, so i "spotted" them asked me where they can enable NAT for the Interface :/ and i sayed
check the box at Masquerade for Zone wlan.

jow-, 2017-05-03:
MSS-Fix is the opposite of MTU Discovery, it is enforcing the interface MTU on packets instead of relying on possible broken pMTU detection.

feckert, 2019-09-19:
related pull request was closed because of inactivity of the pull request creator.

closed 2019-09-19 by feckert