openwrt/luci #1130: "Software Update Lists" fails

notorand-it, 2017-04-28:
Device: TP_Link Archer C7
System Version: LEDE Reboot 17.01.1 r3316-7eb58cf109 / LuCI lede-17.01 branch (git-17.100.70571-29fabe2)
Browser: Google Chrome for Linux v58.0.3029.81 (64-bit), Mozilla Firefox for Linux v53.0 (64 bit)

A. How to reproduce:
1. Reboot the LEDE box and wait for it to go online
2. Select System-->Software menu item
3. Click on Actions-->Update lists

B. Expected result
The software repository index to be updated or downloaded

C. Actual result
Failure page stating:

```
The submitted security token is invalid or already expired!
In order to prevent unauthorized access to the system, your request has been blocked. Click "Continue »" below to return to the previous page.
Continue »
```

D. Tested workarounds
1. logout and close web browser. It doesn't fix.
2. Reboot the device. It doesn't fix
3. Reinstall the whole system. it doesn't fix
4. From command like `opkg update`. It doesn't fix as any other button in the page triggers the same error.

E. Severity
Medium to high, IMHO.

hnyman, 2017-04-28:
Works normally for me in LuCI in 17.01-SNAPSHOT r3326-1ab41265c3 
Firefox 54.0a2 in Windows 10 x64

Have you tried clearing the browser cache?  Or try with "private browsing".

notorand-it, 2017-04-28:
The problem is when you go from OpenWRT 15.05.1 to LEDE 17.01.1.
A complete reinstall (no setting left over) with the "factory" image fixed everything.
Please, close the bug as this needs to be documented in LEDE.

closed 2017-04-28 by hnyman

notorand-it, 2017-05-02:
**UPDATE**
I have the problem persistently on my LEDE 17.01.1 box now.
I thought I had solved the problem by doing a clean reinstall.
I go to Network-->DHCP and DNS-->Static Leases.
I see a "normal" item and an extra empty item below it.
If I try to delete either I get "The submitted security token is invalid or already expired!"
I have deleted all the cache and cookies.
I have rebooted the LEDE box.
I have switched between Firefox and Chrome.
And I still get the error when I click on "Save & Apply" or "Save". But not on "Reset".

crazycat36, 2018-05-30:
I am having this issue with a GL-AR300M with version 17.04. In openwrt it was working. I factory reset, Flashed the firmware again and again both leaving setting and erasing settings. Nothing I've thought of will make this work. All I'm trying to do is add a Wireguard interface. I don't know what i should be doing.

mlavelle42, 2018-06-02:
I'm also having this issue trying to create a new Network/Interface for Zerotier. Submit always returns this error message. Running: Powered by LuCI lede-17.01 branch (git-17.290.79498-d3f0685) / LEDE Reboot 17.01.4 r3560-79f57e422d. Happens in Chrome and Firefox.