openwrt/luci #1143: Guest wifi network setup

fantom-x, 2017-05-01:
It is now a must to have and a lot of consumer routers come with those easily configurable. But it is a challenge to setup in LEDE for new people. I want to add predefined configuration that would include two guest WiFi AP's,

fantom-x, 2017-05-01:
Click something on the phone and submitted without finishing. I will add wireless config, firewall rules, guest firewall zone, etc.

Before I spend any time on this: are there any objections?

deejc-uk, 2017-05-01:
I setup guest today using the recipie. I found on openwrt site but wshaper is missing from packages so guest wifi is a nice idea but without a way to limit bandwidth its not ideal

hnyman, 2017-05-01:
LuCi is probably not the correct place... This is just a GUI.
Neither wifi, network, dhcp nor firewall default configs are defined here. They are in the main LEDE and Openwrt sources.

And why two guest APs? For two radios? Just remember that the default wifi config is created on the fly on the first boot after flash, based on the hardware. Routers have 1,2 or 3 radios. Most supported routers do not even have two radios. (well nowadays maybe the "most" is exaggeration).

(And I am not so sure if a majority of users do create a guest network in their routers.)

fantom-x, 2017-05-01:
Thx for the comments; It looks like I am trying to take on more than I can chew at this time. I will close this request and take some time to think about this issue.

closed 2017-05-01 by fantom-x

megapro17, 2018-02-27:
Open this issue please, very needed ability