openwrt/luci #1148: Status/Overview page displays incorrect memory values

fantom-x, 2017-05-03:
The calculations on the page do not account for the `"cached"` memory

This is what it is on the page:
```
Memory
  Total Available	37704 kB / 479620 kB (7%)
  Free	30992 kB / 479620 kB (6%)
  Buffered	6712 kB / 479620 kB (1%)
```
And this is real values from "top" and "free":
```
top
Mem: 452320K used, 27300K free, 3048K shrd, 6712K buff, 309324K cached
```
```
free
         total       used       free     shared    buffers     cached
Mem:        479620     450352      29268       3048       6712     309324
-/+ buffers/cache:     134316     345304
Swap:            0          0          0
```

jow-, 2017-05-03:
Cached memory is deliberately calculated as available memory by luci as this is memory available for application use.

closed 2017-05-03 by jow-

fantom-x, 2017-05-03:
@jow-, I agree with you that that is the way it should be, but it is currently not. If you look at the figures I provided, total available is reported as 7% by luci, but in "free" it is over 60%: the "cached" portion is not added by the status page.
The page performs "free + buffered" for total memory, but buffered does not include "cached".
What am I missing here?

fantom-x, 2017-05-03:
I forgot to include this info below that is used by the page: "cached" is not included in any of the figures. 

ubus call system info
    {
    	"localtime": 1493765624,
    	"uptime": 5200,
    	"load": [
    		2656,
    		9600,
    		11264
    	],
    	"memory": {
    		"total": 491130880,
    		"free": 33759232,
    		"shared": 3121152,
    		"buffered": 6967296
    	},
    	"swap": {
    		"total": 0,
    		"free": 0
    	}
    }

jow-, 2017-05-03:
LuCI uses the `sysinfo(2)` call to obtain memory information, this call does not provide any "cached" memory information.

fantom-x, 2017-05-03:
Yes, that is a known "feature" of sysinfo. Is there a way to run a shell command from luci? I can work on a fix myself; just need some guidance.

fantom-x, 2017-05-03:
Nvm, I found a way. Do you mind reopening this issue and I will provide a fix?

jow-, 2017-05-03:
Please do not spawn shell commands to obtain that information, the overhead is too bit. We need to find a proper way in C and need to implement it in `rpcd`.

fantom-x, 2017-05-03:
I can open /proc/meminfo and get all the needed values from there and eliminate the use of sysinfo. Would that be acceptable?

jow-, 2017-05-03:
Yes, but it needs to be implemented here: https://git.lede-project.org/?p=project/procd.git;a=blob;f=system.c#l188

fantom-x, 2017-05-03:
That will change the behavior of the existing function: it will start returning different values and might break other things. Doing the change in the page is the safest and backwards compatible. And will be easier to maintain I think.
Sysinfo is a well defined function and I would not want to change it.

jow-, 2017-05-03:
I'll take care of the procd change myself then if I find time. I do not want such one-off parsing code in LuCI as string parsing is very expensive in Lua.

fantom-x, 2017-05-03:
Good point. Are you indeed planning to roll cached into buffered to be returned by sysinfo? 

I want to figure out how to build and run that module, but have not tried yet to set up a build environment. I will see how much progress I can make in that area if I can find good guides.

fantom-x, 2017-05-03:
In case you change your mind about making making the exchange in lua, here is an a example of how I wanted to do it, someone already did that in a fork.

https://github.com/lbthomsen/openwrt-luci/blob/master/modules/niu/luasrc/view/niu/system.htm#L8

ZachCook, 2019-02-11:
[0001-procd-Add-memory2-table-with-Cached-and-MemAvailable.txt](https://github.com/openwrt/luci/files/2849639/0001-procd-Add-memory2-table-with-Cached-and-MemAvailable.txt)

Here is an attempt at adding this functionality to procd, I'm not sure it would be useful to remove the sysinfo call, it has data that would need to be read and converted from /proc/uptime and /proc/loadavg in addition to the data in /proc/meminfo, so I just left the existing stuff as is.

@jow- Is the new blobmsg table (memory2 in my patch) needed to add this data? Or can the new u64's be added to the existing memory table without causing compatibility issues? I will check this myself otherwise, but if you already know how it needs to be done that would save me some time investigating.

I still need to test this out along with luci changes to show the info added, before I send to mailing list, but I want some feedback first on whether this is a good approach.

I'm Planning on changing luci to use "memory2.available" instead of "memory.free + memory.buffered" for the total available memory progressbar, and adding a "memory2.cached" progressbar when these variables are found.

jow-, 2019-02-11:
You can extend the existing table with further blob fields

ZachCook, 2019-02-11:
[luci patch](https://github.com/ZachCook/luci/commit/ffa9584463540a592faf7062594c1fc56b81ca7e)
[procd patch](http://lists.infradead.org/pipermail/openwrt-devel/2019-February/015756.html)

Will send pull request after #2519 is merged, as I based my changes on top of it, and after I have tested it.

ZachCook, 2019-07-08:
@jow- Would you be able to review (and merge if appropriate) [these](http://lists.infradead.org/pipermail/openwrt-devel/2019-February/015762.html) [two](https://github.com/openwrt/luci/pull/2672/commits/688945ef482eb65dfc893d8f8cbc10d0d606c26d) patches? Each has had a quick look over by others, and I have tested them for the past few months on my own router.

commit b436eb05 (2019-11-03): luci-mod-status: use available and cached memory for progress bars

Fixes: #1148

Uses available and cached memory from /proc/meminfo via procd, and carefully
handles the cases where they are nil or zero, so as to not require specific
linux or procd versions to function

MemAvailable is a better estimate than free + buffered/cached, see:
git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=34e431b0

Also adds a new progressbar that shows cached memory if not nil