openwrt/luci #1149: LuCi consistently reports "The submitted security token is invalid or already expired!"

notorand-it, 2017-05-03:
I already opened a bug some time ago about this.
It seemed to me the issue was due to a "dirty" flash from OpenWRT 15.05.1 to LEDE 17.01.1 retaining OpenWRT config over LEDE's.
Now I am getting this error message on almost all buttons I click in the web UI.
I have no knowledge on how to help with logs and so on, but am willing to help.
The web UI is unusable.
This is happening on a Archer C7 with HTTPS enabled.

notorand-it, 2017-05-03:
Older report is [here](https://github.com/openwrt/luci/issues/1130).

crazycat36, 2018-05-30:
I am having this issue with a GL-AR300M with version 17.04. In openwrt it was working. I factory reset, Flashed the firmware again and again both leaving setting and erasing settings. Nothing I've thought of will make this work. All I'm trying to do is add a Wireguard interface. I don't know what i should be doing.

mlavelle42, 2018-06-02:
I'm having this same problem on GL-MT300N-V2. It's a deal killer. Can't create new Interface for Zerotier. Help!

jow-, 2018-06-02:
Which version of LuCI, which browser version? Does it happen via HTTP or HTTPS?

mlavelle42, 2018-06-02:
Powered by LuCI lede-17.01 branch (git-17.290.79498-d3f0685) / LEDE Reboot 17.01.4 r3560-79f57e422d 
Doesn't work in either Firefox and Chrome

RayfenWindspear, 2018-06-06:
Confirming on my GL-AR300M

LEDE Reboot 17.01.4 r3560-79f57e422d / LuCI lede-17.01 branch (git-17.290.79498-d3f0685)

Just noticed that these devices don't let you into the web interface via https :open_mouth: only http

mlavelle42, 2018-06-13:
Where (and what), specifically, did you find the right LEDE? Got a link (that would help me a lot)?

-

RayfenWindspear, 2018-06-13:
You just head to their table of hardware and enter your model number in the input field.

https://openwrt.org/toh/views/toh_fwdownload

Or specifically for your GL-MT300N-V2 @mlavelle42 , which unfortunately looks to be a snapshot instead of a stable release, so YMMV.

https://openwrt.org/toh/views/toh_fwdownload?dataflt%5BVersions*~%5D=v2&dataflt%5BModel*~%5D=MT300N

mlavelle42, 2018-06-16:
I could use your further help with this. I can find instructions for flashing the .bin file but not how to flash lede/openwrt only. Can you provide some advice on this?

awk0324, 2018-09-01:
In the case of GL-AR300M it appears to be an issue in the modified code of Gl-inet's webui. It's also accompanied with an error msg in console saying:
 --HTTP-X-CSRF-TOKEN in /usr/share/glweb/login.html is not transferred
So I would assume this is not an OpenWrt issue. Also I only ran into this issue when creating a new interface in webui. If I for instance create a new wireless interface and let a new network interface be created during the process, this runs fine and it is possible to manage the newly created network iface without any further issues. So this might be a workaround for ppl preferring to stay with the Gl-inet image.

jow-, 2018-10-21:
Seems to be unrelated to OpenWrt.

closed 2018-10-21 by jow-

hex0cter, 2018-12-06:
Does anyone know if there is anyway to work around this? I ran into the same problem on a GL-AR300M. Would appreciate it if someone can provide the equivalent CLI command, as I not familiar with this.

RayfenWindspear, 2018-12-06:
@hex0cter did you try the workaround awk0324 mentioned? I haven't tried it, I bailed and just use vanilla OpenWRT.

hex0cter, 2018-12-06:
@RayfenWindspear I don't think I quite understand what @awk0324 said by `create a new wireless interface and let a new network interface be created during the process`, would you or someone be a little more specific for what needs to be done?

AceAutomation, 2019-05-24:
Greetings evrybody.
I've the same issue. I ran into this issue when creating a new interface in webui. A SimpleForm.
I use MT300N-V2 with OpenWrt SNAPSHOT r9860-9385ff654e / LuCI Master (f138fc93) / Kernel 4.14.111. 
Use snapshot seem not to be a good idea also.
Did you find a solution to this issue since Dec 2018 ?
I tried reinstall luci luci-base luci-mod-admin-full luci-theme-bootstrap
Thanks

The code for your information:
`
local fs  = require "nixio.fs"
local sys = require "luci.sys"
local config = "/etc/config" 
f = SimpleForm("gtw", translate("Configuration File"),
	translate("This is the content of /etc/config."))
t = f:field(TextValue, "gtw")
t.rmempty = true
t.rows = 40
function t.cfgvalue()
	return fs.readfile("/etc/config") or ""
end
function f.handle(self, state, data)
	if state == FORM_VALID then
		if data.gtw then
			fs.writefile("/etc/config", data.gtw:gsub("\r\n", "\n"))
		end
	end
	return true
end
return f
`