openwrt/luci #1152: How does the information from CGI translate other language?

daxiongz, 2017-05-05:
In the html view, I use format as "<%: translate information%>" to come true traanslation.Below is code about it.
[image]
But, how does the information from CGI translate other language?Previous function is useless.
I need to translate English into Chinese.
[image]
Thanks!

musashino205, 2017-05-06:
See [documentation/Templates.md](https://github.com/openwrt/luci/blob/master/documentation/Templates.md).

You can use `translate("Text to translate")`. If variables are included in string, use `translatef("Text to translate with variables %s", var)` instead of `translate("Text to translate") `.

daxiongz, 2017-05-10:
How does the outisde javascript file of luci view page translate other lanuage?Luci self format can't realize.

musashino205, 2017-05-10:
Hmm... I do not know about how to realize with javascript, sorry...

jow-, 2017-05-10:
It seems you attempt to translate code written in C++? The standard solution for that is https://www.gnu.org/software/gettext/, it is also what LuCI's `translate()` was modelled after.

closed 2020-02-23 by jow-