openwrt/luci #1165: firewall editing slow when DNS unavailable

porjo, 2017-05-12:
The following was copied verbatim from [the old bug tracker](http://luci.subsignal.org/trac/ticket/670):

> network -> firewall -> traffic rules -> edit (on any traffic rule)
> 
> it take ages to load
> 
> condition to reproduce:
> set /etc/resolv.conf to "nameserver 192.168.1.222" (any ip that don't exists)
> 
> observation: using tcpdump/strace is showing that luci is doing reverse dns lookup on multiple IPs, then wait for dns timeout

I'm having this issue with recent a LEDE install using luci package `git-17.129.29271-6467df3-1`.

Are dozens of reverse lookups on each rule edit really necessary?

lweller, 2017-06-02:
I have still the same effect with git-17.152.82987-7f6fc16-1, but a checked my dns config and it's slow although everything seams to work fine with nslookup on console.

adelton, 2025-01-10:
I don't experience the UI slowdown with OpenWrt 23.05.5 (r24106-10cc5fcd00). Does the problem persist for you with latest versions?

You say "edit (on any traffic rule)" -- is that any rule which has specific IP address specified, or an rule?

closed 2025-01-10 by systemcrash