openwrt/luci #1170: main index page Active Connections

peterwillcn, 2017-05-20:
http://xxxx/cgi-bin/luci/;stok=xxxx/admin/status/
Active Connections 
```
wc -l /proc/net/nf_conntrack # CPU load is very heavy
cat /proc/sys/net/netfilter/nf_conntrack_count #Please use this. 
```

hnyman, 2017-05-20:
Do you have an example where that code `wc -l /proc/net/nf_conntrack `is currently used in the LuCI master?

That line was recently replaced in master and 17.01 with `/proc/sys/net/netfilter/nf_conntrack_count` by https://github.com/openwrt/luci/commit/7ae244a26ef516511a5e476dd14ec033501a1d8d , so I am not sure where you run into that code in the current LuCI. (naturally it can still be found in some ancient branches)

peterwillcn, 2017-05-21:
Great.
 15.05.1  LuCI (git-16.018.33482-3201903) have this issue

closed 2017-05-21 by peterwillcn