openwrt/luci #1171: output rule not listed accordingly at traffic rules

Cinux90, 2017-05-23:
Hi all,

i face the issue that an output rule is not listed accordingly under Network->firewall-> Traffic Rules.

I use following commands to create an output rule via UCI:
```
uci add firewall rule
uci set firewall.@rule[-1]=rule
uci set firewall.@rule[-1].enabled='1'
uci set firewall.@rule[-1].target='ACCEPT'
uci set firewall.@rule[-1].proto='udp'
uci set firewall.@rule[-1].name='DHCP'
uci set firewall.@rule[-1].dest_port='68'
uci set firewall.@rule[-1].dest='lan'
```

As i check it via iptables its looks as following:
```
Chain zone_lan_output (1 references)
target     prot opt source               destination         
output_lan_rule  all  --  anywhere             anywhere             /* !fw3: user chain for output */
zone_lan_dest_ACCEPT  udp  --  anywhere             anywhere             udp dpt:bootpc /* !fw3: DHCP */
zone_lan_dest_REJECT  all  --  anywhere             anywhere             /* !fw3 */
```

As far as good. But when you looked into the webinterface luci you will see that this rule is listed with "Accept forward". But this is wrong. As seen above its an output rule. Furhtermore its impossible to create an output rule via luci because you can only choose "Any zone" which end with an `firewall.rule.src="'*"` and then its an forward or input rule.

I have face this issue by using the PC Engine APU2C4 with `Linux LEDE 4.9.20 #0 SMP Sat May 13 22:34:04 2017 x86_64 GNU/Linux` self compiledby using this:https://github.com/riptidewave93/LEDE-APU2

vbspam, 2017-12-03:
I can confirm this also on my system.
I see this as not only it show the rule wrong, it is even not possible to create OUTPUT rule with the LUCI gui (rule containing only 'dest' zone).

My system is LEDE "LEDE Reboot 17.01.3 r3533-d0bf257c46 / LuCI lede-17.01 branch (git-17.315.61305-a9bddc4)"

jow-, 2018-06-08:
Fixed with 2eb4015e1 ("luci-app-firewall: recognize egress rules in rule overview")

closed 2018-06-08 by jow-