openwrt/luci #1174: luci-app-asterisk doesn’t do anything with asterisk

nopdotcom, 2017-05-26:
The existence of `luci-app-asterisk` wasted my time, and will waste others’ time.

I was a former user of `luci-app-pbx`, which I didn’t see in LEDE, but there was this tempting `luci-app-asterisk`. It has a reasonable UI. But it just does nothing except generate a UCI config which nothing uses. It is good that it exists, and someday someone will generate config files from it. But it is bad to show it to non-developers.

Please stop shipping this; or at least warn on the luci Asterisk page that configuring Asterisk here does no good. I’ll generate a PR for the warning if you’d like.

chris3081, 2017-06-25:
I have the same issue with asterisk13. I can see its loading commands in the CLI but they don't seem to be setting anything

tits4net, 2018-08-29:
Still the same. It seems the script that generate asterisk configuration file from UCI is missing.

@jow- as you seems to have do some of commit on this project, a long time ago, do you have any idea what is happening ?

Do we have to finish the application or should we remove it ?

FrancYescO, 2018-11-12:
no update on this?

jow-, 2018-11-12:
No, this application was never completed, don't use it. I'll drop it from the repo.

closed 2018-11-12 by jow-