openwrt/luci #1179: LuCI: SSIDs displayed incorrectly if multiple WLANs are configured and at least one is disabled

arminfuerst, 2017-06-01:
I'm sorry, I didn't find a way to always reproduce this error, but I have one device having configured multiple wireless networks where some of them are always disabled. In the overview (/cgi-bin/luci/admin/network/wireless), the SSIDs of the disabled networks are replaced by a SSID of one of the enabled networks.
I realized this bug in git-17.100.70571-29fabe2 (LEDE-17.01.1) and the same config was working properly with OpenWRT 15.05.1.
I also had this error once with a router having just 2 networks and having one of those disabled, but I just tried to reproduce it and wasn't able to.

arminfuerst, 2017-06-27:
I still can reproduce this in git-17.152.82987-7f6fc16 (LEDE-17.01.2).

kubrickfr, 2017-08-12:
[image]

torte71, 2017-12-29:
Same here (17.01.4 for wr842v3 - https://downloads.lede-project.org/releases/17.01.4/targets/ar71xx/generic/lede-17.01.4-ar71xx-generic-tl-wr842n-v3-squashfs-sysupgrade.bin)
Exact steps to reproduce on a clean system:
- flash the router, uncheck "keep settings", confirm (or perform a "reset to defaults")
- log in, goto "password configuration", set password, "save&apply"
- go to "network"/"wireless", enable the given "LEDE" wireless AP (note, that it is displayed as "Mode: Client" in disabled state, but changes to "Master" when enabled)
- press "add" to create a new AP, keep the ESSID "OpenWrt", change mode to "AdHoc", check "lan" in "Interface configuration", "save&apply"
- go to "network"/"wireless" again - note that the new AP displays "Mode: Master" instead of "AdHoc"
- disable the first AP (the "LEDE"/"master" one), wait some seconds: the displayed ESSID will change after ~3 seconds

Now both APs show the same ESSID ("OpenWrt") and mode ("Ad-Hoc")

The detailed "Interface configuration" section at the bottom shows the correct configuration, but the overview displays wrongly.

kubrickfr, 2017-12-29:
Yes, I'm pretty sure this is still happening too on trunk.

jow-, 2018-04-06:
Fixed in master with f9b8d7ff7bbd3c82f9889b65af07a28d94d9fed7

closed 2018-04-06 by jow-