openwrt/luci #1181: wireless (broadcom driver) multiple SSID don't display correctly

yanderemoe, 2017-06-02:
version:` luci - git-17.152.82987-7f6fc16-1`
when I have multiple SSID on broadcom driver (config file below)
`config wifi-device 'wl0'
	option type 'broadcom'
	option txantenna '3'
	option rxantenna '3'
	option hwmode '11g'
	option channel '11'
	option htmode 'HT40'
	option txpower '24'

config wifi-iface
	option device 'wl0'
	option mode 'ap'
	option ssid '**hidden**'
	option isolate '1'
	option encryption 'psk2+ccmp'
	option network 'wifi1'
	option bssid '**hidden**'
	option key '**hidden**'

config wifi-iface
	option device 'wl0'
	option mode 'ap'
	option ssid '*hidden*-guest'
	option isolate '1'
	option encryption 'psk2+ccmp'
	option bssid '**hidden**'
	option network 'wifi2'
	option key '**hidden**'
`
[image]
(displayed same thing)

Doesi, 2017-12-23:
Same here on my ASUS RT-N16 using the proprietary broadcom-driver on 17.01.4.

platkaw, 2018-05-27:
I'm also having this issue on my Asus RT-N16 running OpenWrt SNAPSHOT r7045-c8fdd0e / LuCI Master (git-18.145.51107-9de9a71).

astitcher, 2018-08-22:
I am still seeing this issue with the 18.06.1 release:

OpenWrt 18.06.1 r7258-5eb055306f / LuCI openwrt-18.06 branch (git-18.228.31946-f64b152) 

Running on a Netgear WNR3500L, with architecture BCM4716

astitcher, 2018-08-22:
A seeming side effect is that a single connection to the second SSID is listed twice on the overview page both times attributed to the first SSID

jow-, 2018-08-23:
Please provide the output of `iwinfo` via SSH

astitcher, 2018-08-23:
```
# iwinfo
wl0       ESSID: "George"
          Access Point: 00:90:4C:09:00:01
          Mode: Master  Channel: 1 (2.412 GHz)
          Tx-Power: 32 dBm  Link Quality: 5/5
          Signal: 1 dBm  Noise: -82 dBm
          Bit Rate: 72.0 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: wl  HW Mode(s): 802.11bgn
          Hardware: 04BF:4329 14E4:04CF [Broadcom BCM4329]
          TX power offset: none
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: wl0

wl0-1     ESSID: "Fred"
          Access Point: 02:90:4C:09:00:02
          Mode: Master  Channel: 1 (2.412 GHz)
          Tx-Power: 32 dBm  Link Quality: 5/5
          Signal: -50 dBm  Noise: -82 dBm
          Bit Rate: 72.0 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: wl  HW Mode(s): 802.11bgn
          Hardware: 04BF:4329 14E4:04CF [Broadcom BCM4329]
          TX power offset: none
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: wl0-1

```

jow-, 2018-08-24:
The problem is the lack of runtime information exposed for the proprietary broadcom driver to reliably map config sections to interfaces. The broadcom driver backend will reorder the vif interfaces depending on their type. Do not expect a quick fix for this

astitcher, 2018-11-09:
My issue is purely display, but I'm just using AP mode with multiple SSIDs.

feckert, 2019-10-09:
We haven't heard back from you, so we're going to go ahead and close the issue.
If you're still experiencing the problem, please re-open the issue.

closed 2019-10-09 by feckert