openwrt/luci #1183: uci set ubus command

catiamatos1991, 2017-06-05:
How to do a uci set for config "wireless" section "radio0" option "disabled" value "0" ?

I'm trying to get a new ubus session and the do a uci set and a commit and i got ubus code [0] mas in reality nothing is done to my device cause my set operation command is not working. Anyone can help me?

curl -d '{ "jsonrpc": "2.0", "id": 1, "method": "call", "params": [ "00000000000000000000000000000000", "session", "login", { "username": "root", "password": secret } ] }' -g link
curl -s -d '{"jsonrpc":"2.0","id":1,"method":"call","params":
["'$UBUS_SESSION'","uci","set",{"config":"'teste_wireless'","section":"'radio0'",
"values": { "disabled":1}}]}' -g link
curl -s -d '{"jsonrpc":"2.0","id":1,"method":"call","params":
["'$UBUS_SESSION'","uci","commit",{"config":"'teste_wireless'"}]}' -g
link

hnyman, 2017-06-05:
How many times you are cross-posting this to various forums & bug trackers? So far at least five times today.

closed 2017-06-05 by hnyman

catiamatos1991, 2017-06-05:
Sorry I was trying to get a answer as soon as possible