openwrt/luci #1184: Open Connect VPN compression option incorrectly coded

gavdotcom, 2017-06-06:
LEDE Reboot (17.01.1, r3316-7eb58cf109) 
luci-app-ocserv - git-17.152.82987-7f6fc16-1

/etc/init.d/ocserv seems to be handling the compression option incorrectly. Tick "Enable compression" in luci and /var/etc/ocserv.conf still lists:
#compression = true
Since Open Connect Server defaults to "compression = false", this gives incorrect behavior.

/etc/init.d/ocserv - I changed 
        config_get compression  $1 compression "0"
to:
        config_get compression  $1 compression "1"

...to get expected behavior. Easy fix for someone?

 Also the compression option is not present in /usr/lib/lua/luci/model/cbi/ocserv/user-config.lua whereas other similar options are. I don't know if this is a problem or not.  

Thank you kindly.

gavdotcom, 2017-06-06:
Apologies if this is fixed already in OpenWRT but not LEDE.

hnyman, 2017-06-06:
> Apologies if this is fixed already in OpenWRT but not LEDE.

No need to apologise, as it is the same LuCI.

But looks like you are right, there is a bug.
And the reason is mainly a discrepancy in the main ocserv package, not in the LuCI package.

Compression has been introduced by @nmav to the main ocserv package with https://github.com/openwrt/packages/commit/b83057ce9c069f87816a303d4e8ea6aca8dc612c and to LuCI with https://github.com/openwrt/luci/commit/d0759af923f601585e2515329fc7eb294cb1a0c1

* In LuCI the expected default setting has been set as "enabled", so setting the value in GUI to "enabled" does not cause the value to be written to the config file as LuCI thinks that "no config value" already equals "enabled".
* But in the main package, the default is "enabled" in README example and the ocserv.conf.template example, but not in the ocserv.init init script and no default value has been added to the shipped default /etc/config/ocserv. So the default gets set as "disabled" in practice.

My guess is that the default value "enabled" has been just forgotten to be added to the shipped /etc/config/ocserv.

Maintainer @nmav should decide if the default in the main package is disabled or enabled, and straighten up the main package. Then the LuCI app can be adjusted accordingly (if needed).

Ps. As currently the init script defaults to "disabled" in case no option is set, the default in LuCI might be better set to "disabled". But let's first remove the schizophrenia from the main package.

gavdotcom, 2017-06-06:
Thank you @hnyman for your follow up and insight. I have been reading http://www.infradead.org/ocserv/manual.html  and based on the words that say "Uncomment this to enable compression negotiation (LZS, LZ4)" where the comment is "#compression = true", that implies default is off, and perhaps we should follow that convention?
Thus:
 - No tick in LuCI means off (duh!); means no explicit config statement in /etc/config/ocserv (the opposite of what happens now); means config build script keeps "#compression = true" commented out in /var/etc/ocserv.conf; means a default of off in the ocserv package... and of course the corollary:
- Tick in LuCI means on; means an explicit config statement in /etc/config/ocserv; means config build script uncomments  the "#compression = true" statement in /var/etc/ocserv.conf, and of course this enables compression in ocserv (yay!)

Hopefully @nmav can sort this one out with out feedback. Thanks again.

nmav, 2017-06-06:
thank you guys. I have no time to dig into that sorry. If you submit a merge request which makes sense I'll get it in. If you want to take over maintenance of ocserv on openwrt, I'd be glad too as I unfortunately have no time to spend on it.

rtau, 2018-03-12:
I would like to enable compression on the OpenConnect connections on my little router, which is blocked by this issue.

I have submitted the pull request #1680 which modify the LuCI default to match the ocserv package, as this would not change the setting on existing installations.

Please feel free to comment.

rtau, 2018-09-20:
@gavdotcom, the patch have been merged and should be in OpenWRT 18.06.1, please check if the issue still exists or not.

Thanks

feckert, 2019-09-16:
As I can see in the issue history this issue is fixed so I will close this.

closed 2019-09-16 by feckert