openwrt/luci #1192: access violation luci dispatcher

jkimmel, 2017-06-12:
lime/lede 17.01.2 on wr841 v8
message:
/usr/lib/lua/luci/dispatcher.lua:327: Access Violation
The page at 'admin/system/admin/' has no parent node so the access to this location has been denied.
This is a software bug, please report this message at https://github.com/openwrt/luci/issues
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:327: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>

jow-, 2017-06-12:
Please attach the output if `opkg list_installed`, `cat /etc/board.json` and `fibd /overlay`

jkimmel, 2017-06-13:
root@LiMe-16d948:~# opkg list_installed
alfred - 2017.1-1
base-files - 173-r3435-65eec8bd5f
batctl - 2017.1-0
bmx6 - r2017032301-5
bmx6-auto-gw-mode - 2016-11-07-1478542738
bmx6-json - r2017032301-5
bmx6-sms - r2017032301-5
bmx6-table - r2017032301-5
bmx6-uci-config - r2017032301-5
busybox - 1.25.1-4
bwm-ng - 0.6.1-1
dnsmasq-dhcpv6 - 2.77-1
dnsmasq-distributed-hosts - 2016-09-08-1473363333
dnsmasq-lease-share - 2016-09-08-1473363333
dropbear - 2017.75-1
ebtables - 2015-10-28-4c3e5cd3-1
ethtool - 4.8-1
firewall - 2017-05-27-a4d98aea-1
fstools - 2016-12-04-84b530a7-2
fwtool - 1
hostapd-common - 2016-12-19-ad02e79d-3
ip-tiny - 4.4.0-9
ip6tables - 1.4.21-2
iptables - 1.4.21-2
iputils-ping - 20101006-1
iputils-ping6 - 20101006-1
iw - 4.9-1
iwinfo - 2016-09-21-fd9e17be-1
jshn - 2017-02-24-96305a3c-1
jsonfilter - 2016-07-02-dea067ad-1
kernel - 4.4.71-1-e19b9950ed08c60448d97dc11fbd6537
kmod-batman-adv - 4.4.71+2016.5-4
kmod-br-netfilter - 4.4.71-1
kmod-bridge - 4.4.71-1
kmod-cfg80211 - 4.4.71+2017-01-31-2
kmod-crypto-crc32c - 4.4.71-1
kmod-crypto-hash - 4.4.71-1
kmod-dummy - 4.4.71-1
kmod-ebtables - 4.4.71-1
kmod-eeprom-93cx6 - 4.4.71-1
kmod-gpio-button-hotplug - 4.4.71-2
kmod-ip6-tunnel - 4.4.71-1
kmod-ip6tables - 4.4.71-1
kmod-ipt-conntrack - 4.4.71-1
kmod-ipt-core - 4.4.71-1
kmod-ipt-nat - 4.4.71-1
kmod-iptunnel6 - 4.4.71-1
kmod-leds-gpio - 4.4.71-1
kmod-lib-crc-ccitt - 4.4.71-1
kmod-lib-crc-itu-t - 4.4.71-1
kmod-lib-crc16 - 4.4.71-1
kmod-lib-crc32c - 4.4.71-1
kmod-llc - 4.4.71-1
kmod-mac80211 - 4.4.71+2017-01-31-2
kmod-macvlan - 4.4.71-1
kmod-mt76 - 4.4.71+2017-01-31-3c8caafc-2
kmod-mt76-core - 4.4.71+2017-01-31-3c8caafc-2
kmod-mt7603 - 4.4.71+2017-01-31-3c8caafc-2
kmod-mt76x2 - 4.4.71+2017-01-31-3c8caafc-2
kmod-nf-conntrack - 4.4.71-1
kmod-nf-conntrack6 - 4.4.71-1
kmod-nf-ipt - 4.4.71-1
kmod-nf-ipt6 - 4.4.71-1
kmod-nf-nat - 4.4.71-1
kmod-ppp - 4.4.71-1
kmod-pppoe - 4.4.71-1
kmod-pppox - 4.4.71-1
kmod-rt2800-lib - 4.4.71+2017-01-31-2
kmod-rt2800-mmio - 4.4.71+2017-01-31-2
[... 258 lines cut ...]
/overlay/upper/etc/config/lime-defaults
/overlay/upper/etc/config/firewall
/overlay/upper/etc/config/dropbear
/overlay/upper/etc/config/lime-example
/overlay/upper/etc/config/batman-adv
/overlay/upper/etc/config/luci-bmx6
/overlay/upper/etc/config/alfred
/overlay/upper/etc/config/6relayd
/overlay/upper/etc/config/network
/overlay/upper/etc/config/ucitrack
/overlay/upper/etc/config/system
/overlay/upper/etc/config/uhttpd
/overlay/upper/etc/config/wireless
/overlay/upper/etc/config/libremap
/overlay/upper/etc/ethers
/overlay/upper/etc/dnsmasq.conf
/overlay/upper/etc/firewall.lime.d
/overlay/upper/etc/firewall.lime.d/20-anygw-ebtables
/overlay/upper/etc/profile
/overlay/upper/usr
/overlay/upper/usr/lib
/overlay/upper/usr/lib/lua
/overlay/upper/usr/lib/lua/luci
/overlay/upper/usr/lib/lua/luci/view
/overlay/upper/usr/lib/lua/luci/view/footer.htm

jkimmel, 2017-06-13:
closed by error

jow-, 2017-06-13:
Can you reproduce the problem without LiMe specific packages?

jkimmel, 2017-06-13:
Ich forgot to mention:
Message appears when running  lime-config

antoineclaval, 2017-07-08:
I have the same issue when setting a password on an fresh install. I'm using LiMe-mini, based on Lede 17.01.2, r3435-65eec8bd5f.

opkg is not available there by default. 

I cannot set a password thought the interface, I guess I can do it from the file system or CLI+SSH. but that would be nice to understand what's up.

Thanks

--------- error on http://thisnode.info/cgi-bin/luci/admin/system/admin -------------
/usr/lib/lua/luci/dispatcher.lua:327: Access Violation
The page at 'admin/system/admin/' has no parent node so the access to this location has been denied.
This is a software bug, please report this message at https://github.com/openwrt/luci/issues
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:327: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>

-----------------/etc/board.json---------------------

root@LiMe-c1df1f:~# cat /etc/board.json
{
	"model": {
		"id": "tl-wr741nd-v4",
		"name": "TP-Link TL-WR740N\/ND v5"
	},
	"led": {
		"wan": {
			"name": "WAN",
			"type": "netdev",
			"sysfs": "tp-link:green:wan",
			"device": "eth1",
			"mode": "link tx rx"
		},
		"lan1": {
			"name": "LAN1",
			"type": "switch",
			"sysfs": "tp-link:green:lan1",
			"trigger": "switch0",
			"port_mask": "0x04",
			"speed_mask": ""
		},
		"lan2": {
			"name": "LAN2",
			"type": "switch",
			"sysfs": "tp-link:green:lan2",
			"trigger": "switch0",
			"port_mask": "0x08",
			"speed_mask": ""
		},
		"lan3": {
			"name": "LAN3",
			"type": "switch",
			"sysfs": "tp-link:green:lan3",
			"trigger": "switch0",
			"port_mask": "0x10",
			"speed_mask": ""
		},
		"lan4": {
			"name": "LAN4",
			"type": "switch",
			"sysfs": "tp-link:green:lan4",
			"trigger": "switch0",
			"port_mask": "0x02",
			"speed_mask": ""
		},
		"wlan": {
			"name": "WLAN",
			"type": "trigger",
			"sysfs": "tp-link:green:wlan",
			"trigger": "phy0tpt"
[... 124 lines cut ...]
/overlay/upper/etc/config/luci-bmx6
/overlay/upper/etc/config/alfred
/overlay/upper/etc/config/ubootenv
/overlay/upper/etc/config/6relayd
/overlay/upper/etc/config/network
/overlay/upper/etc/config/ucitrack
/overlay/upper/etc/config/system
/overlay/upper/etc/config/uhttpd
/overlay/upper/etc/config/wireless
/overlay/upper/etc/config/libremap
/overlay/upper/etc/ethers
/overlay/upper/etc/dnsmasq.conf
/overlay/upper/etc/firewall.lime.d
/overlay/upper/etc/firewall.lime.d/20-bmxtun-mtu_fix
/overlay/upper/etc/firewall.lime.d/20-wan-out-masquerade
/overlay/upper/etc/firewall.lime.d/20-bmx-not-over-bat0-ebtables
/overlay/upper/etc/firewall.lime.d/20-anygw-ebtables
/overlay/upper/etc/profile
/overlay/upper/usr
/overlay/upper/usr/lib
/overlay/upper/usr/lib/lua
/overlay/upper/usr/lib/lua/luci
/overlay/upper/usr/lib/lua/luci/view
/overlay/upper/usr/lib/lua/luci/view/footer.htm
----------------------------

rontab68, 2017-12-10:
Same problem with a Dlink 615D router
cat /etc/board.json

{

        "model": {

                "id": "dir-615-d",

                "name": "D-Link DIR-615 D"

        },

        "led": {

                "status": {

                        "name": "Status LED (amber)",

                        "sysfs": "dir-615-d:amber:status",

                        "default": "0"

                },

                "wifi_led": {

                        "name": "wifi",

                        "type": "netdev",

                        "sysfs": "rt2800pci-phy0::radio",

                        "device": "wlan0",

                        "mode": "link tx rx"

                }

        },

        "switch": {

                "rt305x": {

                        "enable": true,

                        "reset": true,

                        "ports": [

                                {

                                        "num": 0,

                                        "role": "lan"

                                },

                                {

                                        "num": 1,

                                        "role": "lan"

                                },

                                {

                                        "num": 2,
[... 303 lines cut ...]
/overlay/upper/etc/firewall.lime.d/20-allow-all-fe80-traffic-over-wan

/overlay/upper/etc/firewall.lime.d/20-bmx-not-over-bat0-ebtables

/overlay/upper/etc/firewall.lime.d/20-anygw-ebtables

/overlay/upper/etc/profile

/overlay/upper/usr

/overlay/upper/usr/lib

/overlay/upper/usr/lib/lua

/overlay/upper/usr/lib/lua/luci

/overlay/upper/usr/lib/lua/luci/view

/overlay/upper/usr/lib/lua/luci/view/footer.htm

/overlay/upper/www

/overlay/upper/www/cgi-bin

/overlay/upper/www/cgi-bin/luci

Powersource, 2018-12-16:
Getting the exact same error, running `lede-17.01.2-lime-mini-ar71xx-generic-tl-mr3020-v1-squashfs-factory.bin`

cshoredaniel, 2018-12-17:
Probably LiME has not been updated to the new split luci-mod-admin-full ... this looks like luci-base is missing the bits from master from the split and reorganisation.  Since LiME is downstream and I was not even aware of it, it wasn't part of that PR.

jow-, 2020-02-23:
The dispatching logic has been completely rewritten in master.

closed 2020-02-23 by jow-