openwrt/luci #1193: Dynamic DNS on the status overview page shows "Config error" under Network.

jebrossard, 2017-06-12:
LuCI lede-17.01 branch (git-17.152.82987-7f6fc16) / LEDE Reboot 17.01.2 r3435-65eec8bd5f

After ADDING a ddns service in LuCI the ddns updating works just fine, but shows "Config error" under Network column on status overview. This is because these options are missing from the newly added service's config:

    option interface
    option ip_source
    option ip_network

Once those options are manually added to the new ddns service's configuration and ddns is restarted, the "config error" goes away. Please see the attached document.  (See issue 1117.)

[config ddns error.docx](https://github.com/openwrt/luci/files/1068218/config.ddns.error.docx)

jow-, 2017-06-12:
@chris5560

jow-, 2017-06-12:
@chris5560 - sorry, you already fixed this issue, it just hasn't been backported to luci-17.01 yet.

commit c79bb964 (2017-06-12): luci-app-ddns: fix to correct system status

preset network "wan", if not configured in ddns config to correct
display system status #1117

[Jo-Philipp Wich: cherry-pick of 3e0492f, fixes #1193]
closed 2017-06-12 by jow-