openwrt/luci #1201: New installation but empty /www

zaros, 2017-06-14:
On OpenWrt 15.05.1 

I installed LuCi using opkg as documentation is written, i.e. commands below:

$ opkg update
$ opkg install luci
$ /etc/init.d/uhttpd start
$ /etc/init.d/uhttpd enable

- I also get a bunch of errors like this after running opkg, in case it's related

* pkg_get_installed_files: Failed to open //usr/lib/opkg/info/libuci-lua.list: No such file or directory.
* pkg_get_installed_files: Failed to open //usr/lib/opkg/info/rpcd.list: No such file or directory.
... etc.

Then I try to load in my browser http://192.168.31.1  

But I only get a directory list and when I check in /www It is empty. How do I get the files?

wwng2333, 2017-06-16:
i think u should check if u router have internet connection

zaros, 2017-06-18:
That's a very strange advice. It obviously does have internet connection
because I did manage to install Luci using opkg.  Am I missing something
here?

wwng2333, 2017-06-18:
maybe u can try this
```bash
opkg remove '*uci*' --force-removal-of-dependent-packages
rm -rf /www/*
opkg install uci luci
uci batch <<-EOF
	set luci.themes.Bootstrap=/luci-static/bootstrap
	commit luci
EOF
/etc/init.d/uhttpd enable
/etc/init.d/uhttpd start
```

feckert, 2019-08-11:
We have now 18.06 and a new 19.07 stable branch 15.05 is not supported anymore I think this is an old issue.

closed 2019-08-11 by feckert