openwrt/luci #1203: status page: show chipset

nschloe, 2017-06-16:
Whenever I need to upgrade OpenWRT/Lede, I'll have to google anew to remember what chipset my router has so I can select the right one from [the target list](https://downloads.lede-project.org/releases/17.01.2/targets/).  It'd be really useful if the status page 
```
Hostname	lede
Model	TP-Link TL-WDR4900 v1
Firmware Version	LEDE Reboot 17.01.1 r3316-7eb58cf109 / LuCI lede-17.01 branch (git-17.100.70571-29fabe2)
Kernel Version	4.4.61
Local Time	Fri Jun 16 15:44:50 2017
Uptime	2d 21h 9m 12s
Load Average	0.00, 0.02, 0.00
```
would list that as well.

bobmseagithub, 2017-07-02:
Hi,
I have 2 lines for you.   Use either or both, depending on how much detail you'd like.
                                                                                                                      
                <tr><td width="33%"><%:Hardware%></td><td><%=luci.sys.exec("grep -i hardware /proc/cpuinfo | cut -f2 -d:") %></td></tr>
                <tr><td width="33%"><%:Hardware%></td><td><%=luci.sys.exec("cat /proc/cpuinfo|  sed 's/$/<br>/'") %></td></tr>

To implement them, edit (on your device) 
/usr/lib/lua/luci/view/admin_status/index.htm 

Around like 686, you'll see:
                <tr><td width="33%"><%:Hostname%></td><td><%=luci.sys.hostname() or "?"%></td></tr>

Then add the 2 lines, for a result like:
                <tr><td width="33%"><%:Hostname%></td><td><%=luci.sys.hostname() or "?"%></td></tr> 
                <tr><td width="33%"><%:Hardware%></td><td><%=luci.sys.exec("grep -i hardware /proc/cpuinfo | cut -f2 -d:") %></td></tr>
                <tr><td width="33%"><%:Hardware%></td><td><%=luci.sys.exec("cat /proc/cpuinfo|  sed 's/$/<br>/'") %></td></tr>

I hope that helps, until the neccesary data is supported directly in Luci.

Thanks, Bob

crackself, 2017-07-16:
@bobmseagithub nice solution! can you tell me what different between  luci-mod-admin-mini andluci-mod-admin-full,thanks.