openwrt/luci #1214: openwrt luci network firewall failed

ytlm, 2017-06-22:
/usr/lib/lua/luci/dispatcher.lua:433: Failed to execute function dispatcher target for entry '/admin/network/firewall'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:433: Failed to execute arcombine dispatcher target for entry '/admin/network/firewall/zones'.
The called action terminated with an exception:
/usr/lib/lua/luci/cbi.lua:311: Unable to read UCI data: firewall
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:433: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:168: in function </usr/lib/lua/luci/dispatcher.lua:167>

ytlm, 2017-06-22:
opkg list_installed | grep luci
luci - git-15.363.78009-956be55-1
luci-app-firewall - git-15.363.78009-956be55-1
luci-app-mwan3 - 1.4-3
luci-app-qos - git-15.363.78009-956be55-1
luci-app-samba - git-15.363.78009-956be55-1
luci-app-upnp - git-15.363.78009-956be55-1
luci-base - git-15.363.78009-956be55-1
luci-i18n-base-en - git-15.363.78009-956be55-1
luci-i18n-base-zh-cn - git-15.363.78009-956be55-1
luci-i18n-firewall-en - git-15.363.78009-956be55-1
luci-i18n-firewall-zh-cn - git-15.363.78009-956be55-1
luci-i18n-qos-en - git-15.363.78009-956be55-1
luci-i18n-qos-zh-cn - git-15.363.78009-956be55-1
luci-i18n-samba-en - git-15.363.78009-956be55-1
luci-i18n-samba-zh-cn - git-15.363.78009-956be55-1
luci-i18n-upnp-en - git-15.363.78009-956be55-1
luci-i18n-upnp-zh-cn - git-15.363.78009-956be55-1
luci-lib-ip - git-15.363.78009-956be55-1
luci-lib-json - git-15.363.78009-956be55-1
luci-lib-nixio - git-15.363.78009-956be55-1
luci-mod-admin-full - git-15.363.78009-956be55-1
luci-mod-rpc - git-15.363.78009-956be55-1
luci-proto-3g - git-15.363.78009-956be55-1
luci-proto-ipv6 - git-15.363.78009-956be55-1
luci-proto-ppp - git-15.363.78009-956be55-1
luci-theme-bootstrap - git-15.363.78009-956be55-1

ytlm, 2017-06-22:
/etc/config/firewall already exists

jow-, 2017-06-22:
Your /etc/config/firewall is corrupt. Run `uci show firewall >/dev/null` for details.

ytlm, 2017-06-23:
$ uci show firewall > /dev/null
uci: Parse error (invalid command) at line 2, byte 0

i cound't understand this error, just learn openwrt about three weeks 0.0.

jow-, 2017-06-23:
It means that your /etc/config/firewall contains a syntax error near line 2. You probably edited it manually with a text editor and introduced a mistake. Either manually fix the error or reset the system to defaults using the `firstboot` command.

In any case this is not a LuCI bug.

closed 2017-06-23 by jow-

ytlm, 2017-06-23:
ok,I know, thank you very much, I will tell the people who write the
firewall.