openwrt/luci #1218: luci-app-statistics: install conflict w/ clamav on libltdl.so

aasgit, 2017-06-25:
Maintainer: @jow-  ?
Environment: armv7l, Linksys WRT3200ACM / LEDE Reboot (17.01.2, r3435-65eec8bd5f)

Description:

If trying to install luci-app-statistcs when ClamAV is already installed this happens (and vice versa: https://github.com/openwrt/packages/issues/4512 )

```
root@lede-wyn:~/scripts# opkg install luci-app-statistics
Installing luci-app-statistics (git-17.170.53336-8955523-1) to root...
Downloading http://downloads.lede-project.org/releases/17.01.2/packages/arm_cortex-a9_vfpv3/luci/luci-app-statistics_git-17.170.53336-8955523-1_all.ipk
Installing libltdl (2.4-2) to root...
Downloading http://downloads.lede-project.org/releases/17.01.2/packages/arm_cortex-a9_vfpv3/base/libltdl_2.4-2_arm_cortex-a9_vfpv3.ipk
Collected errors:
 * check_data_file_clashes: Package libltdl wants to install file /usr/lib/libltdl.so.7
	But that file is already provided by package  * clamav
 * check_data_file_clashes: Package libltdl wants to install file /usr/lib/libltdl.so.7.3.0
	But that file is already provided by package  * clamav
 * opkg_install_cmd: Cannot install package luci-app-statistics.
```

hnyman, 2017-06-25:
I close this as it has nothing to do with LuCI. 

LuCI statistics uses collectd for data collection and that has dependency on libltdl, as does clamav. 
Based on the error messages, the possible clash is between clamav and libltdl.

At the first glance it is a normal dependency, so the clash looks strange. But I have not looked any closer for the installation routine.

closed 2017-06-25 by hnyman

bobmseagithub, 2017-07-07:
Workaround:  
1 -- Delete the links and files that refer to libltdl
2 -- opkg remove clamav
3 -- opkg install clamav

Potential cause:
The clamav package has these file links in it.  
Perhaps it should only have the dependency for libltdl?
I am just learning this package format, so this is just a guess.

Thanks, Bob

The files to be deleted are:
:/usr/lib# ls -la /usr/lib/libltdl.*
lrwxrwxrwx    1 root     root            16 Jun 24 22:33 /usr/lib/libltdl.so.7 -> libltdl.so.7.3.0
-rwxr-xr-x    1 root     root         28681 Feb  5 21:36 /usr/lib/libltdl.so.7.3.0

:/usr/lib# rm /usr/lib/libltdl.*