openwrt/luci #1219: luci_statistics stat-genconfig - should write LoadPlugin logfile earlier in the config file, to support collectd debugging.

bobmseagithub, 2017-06-25:
When stat-genconfig writes a new collectd.conf file, it should write the LoadPlugin section earlier ( or first? ).
Currently, it is written later in the collectd.conf file.

This is to enable debugging of collectd.
Or, maybe there is a way to get full logging out of the current setup?  I don't know what it could be, and would be glad to hear about it.
I know I could do this at the command line, but I need logging after unplanned restarts.

Reproduce:
Install LEDE
Install some collectd plugins ( but NOT the logfile plugin )
install the logfile plugin
/etc/init.d/luci_statistics restart

Actual:
/var/log/collectd.log ( the default logfile location ) is missing messages from plugins loaded earlier in the startup process.

Expected:
messages from all plugins are available in /var/log/collectd.log

Versions:
Hardware: Linksys
LuCI lede-17.01 branch (git-17.152.82987-7f6fc16) / LEDE Reboot 17.01.2 r3435-65eec8bd5f 
base-files - 173-r3435-65eec8bd5f
collectd - 5.5.3-3
collectd-mod-logfile - 5.5.3-3
luci - git-17.152.82987-7f6fc16-1
luci-app-statistics - git-17.170.53336-8955523-1

hnyman, 2017-06-26:
You are probably right at least based on the collectd website:

https://collectd.org/wiki/index.php/Plugin:LogFile
> In order for other plugins to be able to report errors and warnings during initialization, the LogFile plugin should be loaded as one of the first plugins, if not as the first plugin. This means that its LoadPlugin line should be one of the first lines in the configuration file. 

But I am not sure if anything can be easily done, as I think that no mutual priority is defined for the plugins.

hnyman, 2017-06-26:
@bobmseagithub 
Please test the following patch for /usr/bin/stat-genconfig
It seems to work for me (forces the logfile config item to be first plugin in collectd.conf). I have not tested the actual logging any further.

You can edit the change by hand to the last few lines in stat-genconfig. The patch separates logfile plugin from the others and forces it to be handled first.

```
--- /usr/bin/stat-genconfig
+++ /usr/bin/stat-genconfig
@@ -461,8 +461,10 @@

 section("collectd")

+section("logfile")
+
 for plugin in pairs(plugins) do
-       if plugin ~= "collectd" then
+       if (plugin ~= "collectd") and (plugin ~= "logfile") then
                section( plugin )
        end
 end
```

@jow- 
Any obvious faults to the solution?

hnyman, 2017-06-30:
@bobmseagithub 
did you test this?
Just editing the file with nano or any editor editor should be enough.

bobmseagithub, 2017-07-01:
Hi, @hnyman
Thanks for your quick response!
I just got to test this, and it works fine.
One thing... 
I tried to use the Patch command, which failed.  My copy of stat-genconfig seems to have a few more lines than yours did.  I presume this is because we're using slightly different versions of the system.
Again, thanks for the fast ( and effective! ) response.

hnyman, 2017-07-02:
I pushed the fix to master. It no complaints arise, I may later backport it to 17.01 branch

bobmseagithub, 2017-12-02:
No complaints so far.  Is it time to backport to 17.01 branch?

hnyman, 2017-12-02:
backported with https://github.com/openwrt/luci/commit/d2dc32af312097fd9588b8d193308c7d7459a9ca

commit afa7a14c (2017-07-02): luci-app-statistics: set logfile as first item in collectd.conf

Force logfile plugin to be evaluated first so that it gets placed
first in the collectd.conf file.

Reference to https://collectd.org/wiki/index.php/Plugin:LogFile

Fixes #1219
closed 2017-12-02 by hnyman