openwrt/luci #1228: Luci doesn't show mesh_id in 802.11s

bobster-galore, 2017-07-04:
It's possible to set up 802.11s in /etc/config/wireless, Latest Freifunk build Freifunk Berlin Hedy 1.0.0-olsrd0903-alpha 4d4151f r2993+442-b9a408c / LuCI lede-17.01 branch (git-17.152.82987-7f6fc16) worx, only Luci doesn*t show the mesh_id.

kishangondaliya, 2017-09-12:
[Pull request](https://github.com/openwrt/luci/pull/1363) has been submitted.

FreifunkUFO, 2017-10-19:
to the pull request from @kishangondaliya:

is it really needed to set an ESSID when monitor-mode is set?

please have a look to the mesh-communities (f.i. freifunk). 
they are also needed to set the parameter " option mesh_fwding '0'". would be nice to make it configurable via luci!

p.s. freifunk-berlin doenst list that value in corresponding git-issues and their town-specific patch-file-collections. maybe they forgot it to set.

p.p.s. the Name for that 11s-mesh as "802.11s" isnt nice (not in your pullrequest, but same luci-file). Better use "Mesh (11s)". The possibility for translation is also missing at this point.

FreifunkUFO, 2017-10-19:
changes for 11s are also needed for admin->network->wifi
see screenshot: i have some 11s neighbors, but luci shows me a "not assocciätet".

file:///home/um/Schreibtisch/freifunk-11s.png

FreifunkUFO, 2017-10-19:
changes for 11s are also needed for admin->network->wifi
i have some 11s neighbors, but luci shows me a "not assocciätet".

kishangondaliya, 2017-10-23:
@FreifunkUFO ,

- ESSID is optional parameter for monitor-mode, but yes, we can set that.

- For support of mesh parameters, I have list of changeable parameters which could be configured, as soon as I get to merge this changes I will add another pull request for those. I'm not getting any replies from core team ( @jow- @hnyman ) so pull request for mesh ID is still pending.

- Yes, totally agree on 802.11s, we need to give proper readability to this.

bobster-galore, 2017-10-29:
S.a. https://github.com/freifunk-berlin/firmware/commit/dd5f5aa690daba49da791b04cde982916b3f05e4#commitcomment-25259284 this patch makes 802.11s configurable in Luci and working.

bobster-galore, 2017-12-13:
Ok, so far it is possible to configure 802.11s but Luci doesn't show 802.11s correctly in http://router.net/cgi-bin/luci/admin/network/wireless.
It doesn't show correct status, it's deactivated or smthg. Is the status not determinable?
Meshid isn't shown instead of ssid. May be the table-header can be changed to ssid / meshid in second column?

kishangondaliya, 2017-12-13:
My PR https://github.com/openwrt/luci/pull/1363 is still not merged, don't know why they are not giving shit to it even if it working based on your reports.

I will look for wireless status page about status of mesh and will submit PR soon.

bobster-galore, 2018-01-04:
To give an idea what would be the desired configuration in /etc/config/wireless:
```
config wifi-iface
	option device 'radio0'
	option encryption 'none'
	option mode 'mesh'
	option mesh_id 'ff-mesh'
	option network 'wireless1'
	option mesh_fwding '0'
```
in /etc/network it already writes:
```
config interface 'wireless1'
	option _orig_ifname 'radio0.network3'
	option _orig_bridge 'false'
	option proto 'static'
	option ipaddr '6.150.3.1/32'
	option netmask '255.255.255.0'
```
s.a. https://lists.berlin.freifunk.net/pipermail/berlin/2015-May/028157.html for a 2015 test setup with Mesh (11s) and OLSRv2.

FreifunkUFO, 2018-01-08:
@bobster-galore i still cant see any apperance of parameter " option mesh_fwding '0'". (except your config example)

kishangondaliya, 2018-01-08:
@FreifunkUFO none of the mesh params are integrated in LUCI yet. Only you can find is mesh_id which is merged into commit https://github.com/openwrt/luci/commit/267bf83db020295179fad88c6859e50e3406665a.

@bobster-galore can you please close this issue as PR for this is already merged.

closed 2018-01-08 by bobster-galore

SvenRoederer, 2018-01-08:
@FreifunkUFO  see https://github.com/openwrt/luci/pull/1513 for mesh-forwarding

bobster-galore, 2018-01-08:
s.a. correct display Mesh (11s) https://github.com/openwrt/luci/issues/1512
configurable parameters for Mesh (11s)  https://github.com/openwrt/luci/issues/1511