openwrt/luci #1230: luci-app-adblock

build000, 2017-07-05:
when
http://192.168.1.1/cgi-bin/luci/admin/services :
```
/usr/lib/lua/luci/dispatcher.lua:460: Failed to execute firstchild dispatcher target for entry '/admin/services'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:460: Failed to execute firstchild dispatcher target for entry '/admin/services/adblock'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:460: Failed to execute cbi dispatcher target for entry '/admin/services/adblock/tab_from_cbi'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/adblock/overview_tab.lua:10: bad argument #1 to 'parse' (string expected, got nil)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:460: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>
```
- latest update adblock (probably conflict to luci-app-adblock).

dibdot, 2017-07-05:
I've fixed this a few minutes ago ...;-)

build000, 2017-07-05:
heh :dancer:

closed 2017-07-05 by build000