openwrt/luci #1232: wifi hwmode set to 11a on a 11g radio

bobmseagithub, 2017-07-13:
I built a sysupgrade from Snapshot, and flashed it.
The 2.4GHz radio does not work.  The 5GHz radio works fine.
in /etc/config/wireless, the hwmode seems incorrect :

config wifi-device 'radio0'
	option type 'mac80211'
	option path 'mbus@f1000000/mbus@f1000000:pcie-controller@82000000/pci0000:00/0000:00:01.0/0000:01:00.0'
	option hwmode '11a'

Hardware / software:
Model	Linksys Audi (EA3500)
Firmware Version	LEDE Reboot SNAPSHOT r4558-438dcbf / LuCI Master (git-17.192.44320-00ae501)
Kernel Version	4.9.34

I just built this, so I suspect it's something wrong with my build. But, after some study, and a few rebuilds, I can't figure out what it might be. Any advice appreciated.

Thanks, Bob

hnyman, 2017-07-13:
LuCI has no role in wifi detection and creation of the initial /etc/config/wireless, so this is wrong place for the discussion. This is no GUI bug, so I will close this.

LEDE forum would be a better place to seek that kind of help discussion, and then possibly file a bug in the LEDE bug tracker if it really seems like a bug.

But still one hint for you: Are you sure that radio0 is 2.4 GHz and radio1 5GHz? In many routers it is vice versa...

closed 2017-07-13 by hnyman

bobmseagithub, 2017-07-13:
Just in case anyone is interested, here's what I think happened; but I haven't proven it.
. router: linksys EA3500 
. build a sysupgrade.tar - without iwinfo
. install sysupgrade.tar
. note that /etc/config/wireless radio0 has "option hwmode '11a'"

.... fix:
Regenerate Configuration, per: https://lede-project.org/docs/user-guide/wifi_configuration
  rm -f /etc/config/wireless
  wifi config
or: build sysupgrade.tar with iwinfo, delete old /etc/config/wireless, install sysupgrade.tar