openwrt/luci #1234: wifi.getiwinfo returns empty result when called from JsonRpc

alexandertmills, 2017-07-13:
I'm trying to access information about wlan0 though an HTTP POST request to JsonRpc. After successfully authenticating, I made the following request to` [device]/cgi-bin/luci/rpc/sys`
```
{
	"method" : "wifi.getiwinfo",
	"params" : [ "wlan0" ]
}
```
…which resulted in a 200 code and this response:
```
{
	"id": null,
	"result": [],
	"error": null
}
```
I've tried this with user.checkpasswd and a number of other exported libraries which produced expected results and no issues. What's the issue with wifi.getiwinfo?

antscode, 2018-07-30:
I found this blog post about the issue (in Chinese, use translate):

https://blog.csdn.net/jt6562/article/details/21399897

The solution provided by the author worked for me. I SSH'd into my OpenWRT box and modified the following file with the code provided in the article above:

/usr/lib/lua/luci/sys.lua

After a reboot, you can call the alternative **wifi.getiwinfo_rpc** method via json rpc.

jow-, 2018-07-30:
The iwinfo RPC binding has been reworked with 055b621cf5cdc15b24e78408d610a00f6869b8e0 - check the commit message of this commit for further details.

closed 2018-07-30 by jow-