openwrt/luci #1241: [luci-app-openvpn] Can't stop OpenVPN service. It restarts automatically after delay.

Scvairy, 2017-07-21:
Powered by LuCI lede-17.01 branch (git-17.152.82987-7f6fc16) / LEDE Reboot 17.01.2 r3435-65eec8bd5f
luci-app-openvpn | git-17.194.52964-580f012-1
[image]

cugnonr, 2017-07-24:
Same problem on linksys WRT 3200 ACM 
LuCI lede-17.01 branch (git-17.152.82987-7f6fc16) / LEDE Reboot 17.01.2 r3435-65eec8bd5f

Scvairy, 2017-07-25:
I temporary solved the problem by disabling openvpn in init section. I think, it causes automatic restart of the service.

hnyman, 2017-07-25:
What makes you to think that this is a LuCI GUI problem? (instead of a problem in the package itself)

Wedmer, 2017-07-27:
All start/stop functionality is made by using service init script. When you stop service it could take some time to finish as it can have some unfinished tasks.

Weishaupt, 2017-11-10:
I'm running LEDE 17.01.4 and I expericence the same behaviour. Stopping via `/etc/init.d/openvpn stop` will actually disconnect OpenVPN. When I use the luci app and try to disconnect, it will start a new process shortly after tearing down the old one.

Wedmer, 2017-11-11:
`/etc/init.d/openvpn stop` stops all openvpn instances.
Please check if `/etc/init.d/openvpn stop name_of_the_instance` stops only one instance.

Weishaupt, 2017-11-11:
There is no such command. `/etc/init.d/openvpn stop name_of_the_instance` gives me 
> Command failed: Not found

The service description also doesn't mention starting / stopping just an instance

> Syntax: /etc/init.d/openvpn [command]
>
> 
> Available commands:
>         start   Start the service
>         stop    Stop the service
>         restart Restart the service
>         reload  Reload configuration files (or restart if that fails)
>         enable  Enable service autostart
>         disable Disable service autostart
>

blamaz, 2017-11-28:
I'm running Davidc502 fork snapshot r5422 and have the exact same behaviour.

Wedmer, 2017-11-30:
There're no support for stopping/starting single instance in current openvpn init script. When I have more free time I'll look for workaround.

dibdot, 2018-09-16:
Not a LuCI issue at all.

closed 2018-09-16 by dibdot