openwrt/luci #1253: luci-app-ddns unnecessarily requires <query> in url string

james-knippes, 2017-07-26:
Procedure:
set up an DDNS provider as "-custom-" provider and filling out the "cutom update-url" field like this:
`http://some.domain./RANDOMTOKEN123/`

Result:
`get the error: "openwrt Custom update-URL: <QUERY> missing / required"`
and can't finfish the setup.

Explanation:
luci-app-ddns expects the update-url to contain a "?"-symbol as inicator for a query. but some ddns-services don't use the query part for updating the ip. 

Expected Result:
can finish the setup even without a "?" in the update-url.

Workaround:
just put a "?" at the end of the update-url, as it shouldn't matter. so it looks like:
`http://some.domain./RANDOMTOKEN123/?`

affected dyndns service: afraid.org

chris5560, 2017-07-27:
From my knowledge the update-URL for afraid.org using "keyauth" (Token) is
http(s)://freedns.afraid.org/dynamic/update.php?[PASSWORD]&address=[IP]
where [PASSWORD] is replaced by the token and [IP] by the current IP of the box.

"afraid.org-keyauth" is already part of the current services/services_ipv6 files
as part of current ddns-scripts package. So no need to use -custom- settings.

Did afraid.org changes it's update-URLs ?

james-knippes, 2017-07-27:
freedns has four diferent mechanisms.
the one you were probably refering to was probably "Username and Password, inline":
http://sync.afraid.org/u/?u=[USERNAME]&p=[PASSWORD]&h=[DOMAIN]&ip=[IP]
i didn't test this method. also i dont know which of the ddns-script options is its qeuvalent.

the probably most used/convinient method is the "Randomized Update Token"
which looks like this:
http://sync.afraid.org/u/R4nd0MT0KEN/
it's probably the most used one, as it is the default one and doesnt require username/password in the url.

apart from that. It doesnt make much sense to force a "?" charater in the URL for the "-custom-" config, as it should be the most flexible one i think.

chris5560, 2017-07-29:
How do you force the IP to set for your configuration, if you don't like to use autodetect ?
Does afraid.org change it's update host and update-URL ?
I cannot find any information on their homepage http://freedns.afraid.org/scripts/freedns.clients.php looking for OpenWRT
Please support me to also update ddns-scripts.

james-knippes, 2017-07-29:
I'm not sure if I completely understand your question. With the randomized Token there is no IP you can set. It's autodetected at the afraid server. With the other methods e.g. "username and password inline" you can set the IP in the URL.

I think all the old update hosts are still valid. But they also have new ones. 
So it seems afraid.org has two major versions for their dynamic dns interface.

The first (old) Version supports a randomized URL with nothing to configure. It looks like this:
`https://freedns.afraid.org/dynamic/update.php?xxxxxxxxxxxr4nd0mT0kenxxxxxxx=`
and a second (old) method is this:
`http://[USERNAME]:[PASSWORD]@freedns.afraid.org/nic/update?hostname=[DOMAIN]&myip=[IP]`

Then there are four "new" Methods in the "updated iterface v2":
Method 1: "Randomized Update Token"
`http://sync.afraid.org/u/r4nd0mT0ken/`
Method 2: "Username and Password, inline" ( IP may be optional? , hadn't time to test it )
`http://sync.afraid.org/u/?u=[USERNAME]&p=[PASSWORD&h=[DOMAIN]&ip=[IP]`
Method 3: "Username and Password"
`http://[USERNAME]:[PASSWORD]@sync.afraid.org/u/?h=[DOMAIN]&ip=[IP]`
Method 4: "/nic/update"
`http://[USERNAME]:[PASSWORD]@sync.afraid.org/nic/update?hostname=[DOMAIN]&myip=[IP]`

That are all methods i found on their website.
Just ask if you need aditional information/help.

james-knippes, 2017-07-29:
It seems someone already added some support for the new Interface two weeks ago.
https://github.com/openwrt/packages/commit/251e5c9cb1a2a82d16327935ef2830f9abf8a8d2
he uses the v6.sync.afraid.org subdomain for ipv6 updates. I didn't find where this is documented...

feckert, 2019-08-11:
I think this is fixed

closed 2019-08-11 by feckert

vollmann-ariel, 2021-01-16:
It's still not working.
I did exactly as @james-knippes did but the ddns server did not update my IP.
Finally I found a workaround to solve it:
In advanced settings I selected as IP address source the option URL and I put the URL "http://sync.afraid.org/u/R4nd0MT0KEN/" in the "URL to detect [IPv4]" text field.
It finally worked, but is not a neat solution at all.

Thanks guys.
[image]