openwrt/luci #1255: Fail to add multiple SSID on Raspberry Pi3

edmondhk, 2017-07-28:
Model | Raspberry Pi 3 Model B Rev 1.2
Firmware Version | LEDE Reboot 17.01.2 r3435-65eec8bd5f / 			LuCI lede-17.01 branch (git-17.152.82987-7f6fc16)
Kernel Version | 4.4.71

I start with connect to my AP, as wwan, everything fine, can ping google.
Then I try to add another SSID by webgui, WPA2&Open wifi.
It seems added to the list but link to internet loss, can't search for LEDE by wifi neither
UART keep give out following error. 
Details kernel log attached
[kernal log.txt](https://github.com/openwrt/luci/files/1182012/kernal.log.txt)

 [  606.269497] brcmfmac: brcmf_update_bss_info: wl dtim_assoc failed (-23)
[  606.279933] ------------[ cut here ]------------
[  606.288290] WARNING: CPU: 2 PID: 74 at compat-wireless-2017-01-31/net/wireless/sme.c:752 0x7f16fbbc [cfg80211@7f14a000+0x37c90]()
[  606.304088] Modules linked in: pppoe ppp_async iptable_nat brcmfmac pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT usbhid slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables hid_generic crc_ccitt compat brcmutil snd_bcm2835 hid evdev ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables snd_pcm_dmaengine snd_compress snd_pcm_oss snd_mixer_oss snd_pcm snd_timer snd_rawmidi snd_seq_device snd_hwdep snd soundcore nls_utf8 vfat fat nls_iso8859_1 nls_cp437
[  606.405865] CPU: 2 PID: 74 Comm: kworker/u8:2 Tainted: G        W       4.4.71 #0
[  606.418542] Hardware name: BCM2709
[  606.426897] Workqueue: cfg80211 0x7f14a11c [cfg80211@7f14a000+0x37c90]
[  606.438369] Function entered at [<800203d4>] from [<8001cd58>]
[  606.448972] Function entered at [<8001cd58>] from [<801ad550>]
[  606.459347] Function entered at [<801ad550>] from [<800277cc>]
[  606.469499] Function entered at [<800277cc>] from [<80027884>]
[  606.479423] Function entered at [<80027884>] from [<7f16fbbc>]
[  606.489127] Function entered at [<7f16fbbc>] from [<7f14ecb4>]
[  606.498723] Function entered at [<7f14ecb4>] from [<7f14edac>]
[  606.508237] Function entered at [<7f14edac>] from [<7f14a130>]
[  606.517667] Function entered at [<7f14a130>] from [<8003c630>]
[  606.527014] Function entered at [<8003c630>] from [<8003cad4>]
[  606.536355] Function entered at [<8003cad4>] from [<80041b50>]
[  606.545693] Function entered at [<80041b50>] from [<80009c38>]
[  606.555058] ---[ end trace e726c8ce9a554a41 ]---

moham96, 2017-12-02:
same problem  here

feckert, 2019-08-02:
Is this still an issue on 18.06 and 19.07 ?

closed 2019-08-11 by feckert