openwrt/luci #1258: LuCI overrides option ttylogin '1' in /etc/config/system

ffries, 2017-07-28:
In a production environment, option:
ttylogin '1'
allows to set a password to serial console.

It will stop some attackers, but not all of them.

However, connecting to LuCI:
/cgi-bin/luci/admin/system/system

will override  completely ttylogin settings.
Then the OpenWRT/LEDE can be hacked easily.

There should either be:
* Support for ttylogin in LuCI
or
* LuCI should preserve ttylogin settings.

Kind regards,

jow-, 2017-07-28:
I cannot reproduce the problem. Added `option ttylogin 1` to my /etc/config/system, navigated to ` /cgi-bin/luci/admin/system/system`, changed some settings, ran Save & Apply.

The `ttylogin` option was preserved and not touched by LuCI.

closed 2017-07-28 by jow-

ffries, 2017-08-08:
I agree, the bug is gone on my firewalls too.
Will submit a patch to add option ttylogin 1 to luci-mod-admin-full.