openwrt/luci #1269: SAMBA give partial sharing...

Skyfinder84, 2017-07-31:
Device: Zidoo-X8
LUCI VERSION Powered by LuCI QA160627-nuplayer_Kylin_2017-05-11_SQA_Dailybuild_NAS Release (git-17.130.47916-3e80cb1) / OpenWrt Chaos Calmer 15.05.1
Zidoo Android ambient is updated
Kodi 17 Kripton installed...

SAMBA IS ACTIVATED ON:
Openwrt options
Device advanced options...
Kodi options...

STRUCTURE:
- WD 4TB linked via USB 3.0 to the Device.
- Device on lan via ethernet...
- Asus Modem / Router... no bullshits about port forwarding, ip... all auto...
- Desktop: Win10 machine all regular, no crack..on in LAN.

The Workgroup is the same for everyone.

THE ISSUE:
When i connect via WIN10 to the HDD...i can see everything.

Skyfinder84, 2017-09-02:
WHY NO ONE TRIES TO HELP???

LipkeGu, 2018-01-18:
Samba share acl depends on underlaying filesystem acls, so you should ```chmod``` the folder via ssh... and restart samba.

feckert, 2019-08-11:
That issue is for 15.05 which is EOL. And I think this is not a LuCI related issue.

closed 2019-08-11 by feckert