openwrt/luci #1286: Firewall logging of rejected packets

ffries, 2017-08-07:
Per discussion on LEDE forum:
https://forum.lede-project.org/t/logging-dropped-rejected-packets-to-syslog-using-uci

uci/LuCI allows logging of rejected packets per zone.
If would be nice to have a general logging feature available in General settings:
luci/admin/network/firewall
[x] Log rejected packets in the following zones:
[ ] Wan
[ ] Lan

Logging should be disabled as always.

jow-, 2017-08-07:
That is redundant to the per-zone logging options already available in the advanced tab.

closed 2017-08-07 by jow-

ffries, 2017-08-08:
I understand the logic, but please reconsider your choice.
Logging is not an "advanced feature".

Logging is a very important security consideration.
maybe the most important issue after choosing which ports to open/forward/close!

i would prefer to see logging in Zone settings, just below "Input", "Output" and "Forward".
For a user, it means "What do we do for those rejected packets?".